sinara-hw/Thermostat #2: project review

gkasprow, 2018-08-18:
@hartytp @jordens I sketched first release.
Please have a look at it.
I also updated wiki with drawings.

jordens, 2018-08-20:
Nice!
1. I'd hook up the ENABLE pins to the uC plus pulldowns and alternatively 0R to GND.
2. You might want to check the output capacitor of the PoE module. They have 470µF there.
3. We need to check that we meet maintain power signature on PoE. That's a couple hundred mW mimimum power consumption IIRC. We should also avoid the acoustic noise the PoE module DS talks about when it runs at low power.
4. Isn't it more efficient to send 5V (instead of 12V) to the other board with the extension channels? The step-down module can handle it and the PoE module as well. You could even use one of those board-to-board edge connectors that you found for Sensor-Box.
5. Can we hook up the LED outputs of the wiznet module to the RJ45 jack LEDs as well?
6. Any chance we can connect pins 15 and 16 (I2C) of the Zotino-TEC connectors to the µC?
9. I'd explicitly pull up RESET of the wiznet module.
7. Congrats to squeezing it on two layers!

hartytp, 2018-08-20:
@gkasprow Great! Travelling atm but will probably be able to look at this in the next few days.

gkasprow, 2018-08-24:
4-channel extension lets you control second module equipped with just TEC controllers from first module with Ethernet, PoE and DC/DC. I'm a bit afraid of exposing 5V to the panel. Someone may plug 12V supply by chance since other EEMs offer such ability.

hartytp, 2018-08-24:
> 4-channel extension lets you control second module equipped with just TEC controllers from first module with Ethernet, PoE and DC/DC.

Okay, makes sense. Have you designed this board yet? Is it worth doing this? IMHO, if people need 4 channels, it's easier to just have two identical thermostat cards with separate ethernet/PoE connections, instead of having to have two separate (non-identical) boards. So, I'd be tempted to scrap this, I don't feel strongly about it.

Also, is there enough power from the PoE module to run 4 TECs safely?

> I'm a bit afraid of exposing 5V to the panel. Someone may plug 12V supply by chance since other EEMs offer such ability.

I meant removing the barrel connector completely and only supporting PoE (I thought this is what we had agreed before). PoE injectors are pretty cheap, so I don't see much need to support both PoE and barrel connectors.

gkasprow, 2018-08-24:
@hartytp It is the same board, but without PoE and Ethernet module.
PoE is rated for 35W

hartytp, 2018-08-24:
sounds good then.

One final request: could you add a comment on the schematic with the p/n (and mating p/n) for those din connectors, please? That way I don't have to load the alltium files to figure it out.

hartytp, 2018-10-09:
Good find. If you use them, let me know how they work out.

gkasprow, 2018-10-09:
@jordens I use them in the lab. I already posted [pics ](https://github.com/sinara-hw/sinara/issues/188#issuecomment-327180612) with them some time ago :) They are really nice.

hartytp, 2018-10-09:
@gkasprow did you ever produce any Thermostat prototypes? I'd like some if they're going :)

gkasprow, 2018-10-09:
Not yet, I'm collecting designs to spare on NRE costs. I will produce it together with Banker extension boards.

hartytp, 2018-10-09:
Good plan. I'm not in a massive hurry for this, but it would be nice to have at some point.

hartytp, 2018-10-09:
Closing this issue since I think the design review is not complete

closed 2018-10-09 by hartytp