sinara-hw/Thermostat #3: Firmware

wobucho, 2018-10-24:
Hello everyone,

as a student of @gkasprow I have made firmware for thermostat's TM4C chip and I want to receive a feedback from you. Source files are in firmware branch. I will try to describe features of this project.

I decided to use all options provided by producent of MTD415T. That is why I made as many as 12 subpages dedicated for handling of MTD415T. For each subpage you have several parameters displayed on the screen. After every reload of subpage T4C asks MTD415T for current parameters through UART. I mention this, because it is important in case of possible changes of subpages. For instance, temperature parameters have separate subpage for each channel. They can be joined together in order to present all parameters for all channels in one subpage, but during subpage loading there would be more commands sent to MTD415T.

The main page includes brief info about project. On the left side you have menu.

[image]

"Temperature status" subpage is made for quick checkout of temperature parameters. Here you have preview of set and actual temperature for two channels.

[image]

"Connection status" subpage presents all connection parameters of 2 channels.

[image]

Then we have two subpages "Port settings" for two channels. There is possibility to change telnet settings here. Although it looks like you can change UART settings, each drop-down list for UART has only one argument. This is because MTD415T accepts only this set of UART parameters.

[image]

"TC general info" shows factory info for two TEC drivers and allows reading and resetting error registers of these two elements. Readout of error register is shown in corresponding row. After resetting some ER, there is short info which register was reset on a reloaded page.

[image]

Then we have 6 pages, 3 for each channel, where one can change parameters of MTD415T. Each changeable parameter has dedicated text field, where one can put new value. Next to each input field there is "Set" button. If setting value ends correctly, we return to the same page (reload).

[image]
[image]
[image]

Each updated value need to comply with requirements described in MTD415T datasheet 
( https://www.thorlabs.com/drawings/a1dca064a8c36849-5FD63D97-0CC3-7E3B-93215C8CFFBDCF31/MTD415T-DataSheet.pdf pp. 12-16). In case of temperature, input value should be given in degrees Celsius, with "." or "," as a decimal separator.
 If setting value ends incorrectly (eg. wrong value - letters instead of numbers, value not in range or timeout of MTD415T), you get "perror.shtm" page, which informs you that some error happened.

[image]

I have set timeout of circa 2 seconds for setting a value. If T4C would have problem to communicate with MTD415T, you will get "timeout" instead of value.

[image]

There is also possibility to communicate with MTD415T directly through Telnet. You can see default parameters of telnet in the printscreen of "Connection Status" subpage. Each character sent via telnet is simply transmitted to corresponding MTD415T.

Besides two UART channels for two MTD415T, it is also a third channel for debugging. It displays info about project during launch and also IP address when connection is established. In case of connection errors, it also shows info.

As for possible modifications that came to my mind, I suppose it would be comfy to have ranges of acceptable values somewhere close to input fields, but I don't know if it would be helpful in case of all parameters.

Anyway, I am open to your suggestions.

Wojtek Obuchowicz

gkasprow, 2018-10-24:
@hartytp @jordens @dtcallcock  @jbqubit @cjbe @sbourdeauducq please have a look at the project of my student.

hartytp, 2018-10-25:
@wobucho cool! Thanks. I'll have a play with it when I get my hands on Thermostat.

I haven't really looked at the code yet. Apart from to notice that you've used c and FreeRTOS. I don't mind that so long as the code is clean, well written and well documents. Which, for a very brief skim, it appears to be. However, others might have different preferences (e.g. https://github.com/m-labs/ionpak).

One thing I that is important to me is to have a very simple SCPI-like interface over TCP. e.g that allows me do write a python program that logs the temperature and emails me if there is an issue. Did you implement this yet? The SCPI interface needs to be well-documented and (as well as the web interface) perform proper input validation etc (e.g. gaurd against the usual things like a negative number setting the temperature to max).

>Although it looks like you can change UART settings, each drop-down list for UART has only one argument. This is because MTD415T accepts only this set of UART parameters.

Why have drop down menus for them? If they can't be changed then I wouldn't even bother displaying them (the user doesn't need to know about them).

Anyway, it's hard to give more detailed feedback without actually trying Thermostat out myself.

hartytp, 2018-10-25:
@gkasprow one other thing about this: the motivation for using those Thorlabs chips instead of designing our own (nicer) circuit was mainly that we didn't have the resources to develop the software. The hardware design we already have based on a 24-bit ADC + IC peltier driver works really well, and could be integrated into Thermostat very quickly. However, it needs software to make it work.

So, if you have a student interested in developing the software for this then we could consider designing Thermostat v2.0 with a nicer circuit.

Anyway, just a thought.

dtcallcock, 2018-10-25:
> One thing I that is important to me is to have a very simple SCPI-like interface over TCP. e.g that allows me do write a python program that logs the temperature and emails me if there is an issue. 

This would definitely be useful.  One issue I'm finding as we incorporate more of this kind of hardware though is that we have a lot of fairly dumb Python programs running in order to poll a devices like this and write the result to our InfluxDB database.  It would be cleaner and more robust if this device could be set up to just log to the database directly (then we can handle alerts centrally in Grafana).  It would just need a page where you can enter your server details and logging preferences (what to log and how often etc.).  Would anyone else find that useful?

> it would be comfy to have ranges of acceptable values somewhere close to input fields

Yes, as this is a webpage, can there be a condensed version of the documentation appended to the bottom of the page so you don't have to look up the datasheet every time you want to do something?

hartytp, 2018-10-25:
> One issue I'm finding as we incorporate more of this kind of hardware though is that we have a lot of fairly dumb Python programs running in order to poll a devices like this and write the result to [...]

Personally, I try to keep the firmware as simple as possible and avoid adding this kind of feature to it. Different users might use different data bases or different modes of logging, so nice not to bake that into the fw. Also, if there is any change to the db protocol you don't want to have to rebuild the firmware. Simpler firmware is also less likely to crash randomly, which is always a bonus.

I don't mind having a bunch of simple python scripts that run on some PC at statup and push data to influx db. It's more flexible and easier to keep up to date/fix bugs.

dtcallcock, 2018-10-25:
>  Also, if there is any change to the db protocol you don't want to have to rebuild the firmware.

Isn't this running FreeRTOS?  I don't know much about that OS, but shouldn't it be able to load a simple 10-line logging script without having to be recompiled and deal with any errors it throws?  It also doesn't seem like it's adding a lot of additional complexity given this thing can already serve up web pages etc.

> Different users might use different data bases

But Artiq specifically uses InfluxDB.

jordens, 2018-11-07:
@wobucho A couple comments.

* You should post your code for review. Just screenshots don't mean much.
* Simplify the web interface as much as possible. Style, colors, and logos don't matter nearly as much as technical soundness, robustness, maintainability, and functionality. Use a templating system to help you with maintaining code vs pages separation and to allow you to reuse code for other interfaces to the API.
* Please try to make your code is neat, compact, and adherent to some consistent style. I'd be happy to get reduced functionality if the rest is of sufficiently high code quality to be reused.
* Don't implement functionality that already exists in the temperature controller module. If e.g. it checks for errors in parameters or checks parameter ranges and tells you, then let it do that and don't try to duplicate the functionality.
* Put focus on automatic testing of all functionality. Setup a CI for the code. That enables both smooth development of your software as well as testing in the factory. It also makes contributions by others easier. Do that early.
* Make sure the remote interface is a prime target. I'd go for MQTT. It's portable and backend-agnostic, meshes seamlessly with all the existing tooling that we have (including influxdb/telegraf), and is a perfect match for the embedded device. Use mDNS for broker discovery as default. If you don't want to or can't, add some SCPI-like interface. But that means triple the effort for the end user.
* Make sure people can download all settings and upload them again in bulk.
* Consider looking at OTA firmware upgrades. Expose the version/revision of your software on all API (web and MQTT/text).

hartytp, 2018-11-08:
@wobucho The other big thing is to make sure this is really reliable. e.g. consider using a watchdog so that if the FW crashes it reboots into a sane state rather than baking the load, etc.

The "killer features" for me would be:
1. Decent ways of plotting the temperature, heater current, etc
2. a half-decent auto-tuner

It probably makes most sense to do those via a python script that connects to thermostat via the remote interface

jbqubit, 2018-11-08:
> Decent ways of plotting the temperature, heater current, etc

Web interface is nice. Proper integration with ARTIQ should also include NDSP and influx logging via artiq_influxdb.py.

hartytp, 2019-01-29:
closing as this is relevant to the Thorlabs version of thermostat, not this version.

closed 2019-01-29 by hartytp