sinara-hw/Thermostat #4: thermostat v2.0

hartytp, 2018-11-07:
We've had some problems recently with a common commercially available temperature controller, which we've tracked down to inadequate EMI filtering in the temperature controller AFE and use of unscreened cabling in the wiring harness provided by a different vendor (the 10k NTC has a 100uA sense current so is a high-impedance node which is very prone to pickup even at 50Hz). 

To solve this, I'd like to swap out these temperature control units for Thermostat, but there are two annoyances:
1. The specified temp co of the Thorlabs chip is quite high and likely to be problematic if we don't temperature stabilise Thermostat, which is a pain
2. We use a resistive heater rather than a TEC for this project. AFAICT there is no software option to make the Thorlabs chips run unipolar (AFAICT the designers didn't consider resistive heaters as an option so didn't bother to implement this in firmware). So, if any transient reduces the heater current below 0 then it will go into thermal run away.

So, I'd like to revive the discussion about replacing the Thorlabs chip with something along the lines of @cjbe's design based on a 24-bit ADC and the Maxim TEC driver IC. As a reminder, the big win from using a 24-bit sigma-delta ADC is that the sample rate is high and the ADC includes very good digital filtering (including extreme rejection of 50/60Hz signals). Combined with a basic CM/DM analog RFI filter, this provides a level of EMI rejection that is essentially impossible to achieve in the analog designs that seem to be used in almost all COTS units. Using @cjbe's prototypes of this design, we were able to achieve stabilities better than 100uK long term without issue.

As I see it, the main challenge here is software. The hardware redesign is very easy and I'd be happy to do that myself using @gkasprow's Altium library. The more involved part is writing decent software: command handler; simple PI(D?) loop; some basic features like thermal runaway detection, current limiting, anti-wind up, etc. I don't have time to do that part. @gkasprow do you have a student who might be interested in doing that? (otherwise, if anyone is potentially interested in funding M-Labs to do this then let us know!)

gkasprow, 2018-11-07:
@hartytp I can add AD7191 to existing design. @cjbe how did you couple the PIC processor with  MAX1968 CTLI input? Using PWM?

gkasprow, 2018-11-07:
We can ask the student to port the PIC code to the TI CPU.

hartytp, 2018-11-07:
> We can ask the student to port the PIC code to the TI CPU.

Which PIC code do you mean?

gkasprow, 2018-11-07:
https://github.com/sinara-hw/Thermostat/issues/5#issuecomment-357445261

cjbe, 2018-11-07:
@gkasprow yes - I generate the CTLI signal from a PWM output and a two-pole reconstruction filter (10k,100n).

hartytp, 2018-11-07:
@gkasprow perfect! I agree: these should be two separate designs rather than trying to fit everything on one board. The Thorlabs version gives us something we can use with minimal effort/risk of hardware bugs, the ADC version should work better in the long run but will need software development and careful hardware characterisation.

gkasprow, 2018-11-07:
@hartytp please have a look
[Thermostat.PDF](https://github.com/sinara-hw/Thermostat/files/2559336/Thermostat.PDF)

hartytp, 2018-11-08:
@gkasprow wow! Thanks for turning that around so fast!

@cjbe any thoughts? Things you'd want to change if you did this design again?

Basically that looks good to go as far as I'm concerned. Small thoughts/comments:
- Cosmetics could be improved but that's non-crucial e.g. would be nice to neaten things up a little, add a power budget, etc. (e.g. I find the fact that the signal flow is R->L on the 3v3->vcc->vref block on the adc makes things a bit hard to follow) 
- @cjbe's design is a bit long in the tooth now and there are newer better ADCs around. However, based on his design notes, it seems that the ADC isn't really a limitation here. The only way I can see where the ADC comes close to being a limitation is the +-1ppm/K gain drift. The best ADCs on the market are still only a factor of two better than that, so the overall improvement is relatively small. Probably not worth the hassle of switching ADCs.
- what are the inductors on the TEC driver? Let's use ones that retain their inductance at 1.5A and pick ones that won't radiate like crazy!

jordens, 2018-11-08:
As a humorous side remark: I've heard talks by high profile representatives of well known companies in all seriousness claiming that inductors are what's limiting miniaturization, integration, scaling, and technological progress in laser current and TEC drivers.

hartytp, 2018-11-08:
@gkasprow what capacitors (voltage rating, case size, dielectric) are you using for IC1? That reg needs 1uF for stability, so we need to make sure we don't use a cheapo ceramic which gives far less capacitance when biassed.

gkasprow, 2018-11-08:
You won't find 1uF NPO. I used X7R.

hartytp, 2018-11-08:
Yes, X7R is good for this, but we need to watch the case size and voltage rating or the capacitance can be way lower than expected. Since this is required for stability, I'd probably make it a slightly higher capacitance and use a decent voltage rating and case size.

hartytp, 2018-11-08:
Can you add an annotation that the 5k11 resistors must be 5ppm/K temp co?

(if you upload the altium files I'm happy to do this).

hartytp, 2018-11-08:
@gkasprow let's tie the ADC CLKSEL high to ensure the internal clock is correctly enabled.

hartytp, 2018-11-08:
let's connect the ADC TEMP pin to the microprocessor to allow access to the ADC's on-board temperature sensor.

hartytp, 2018-11-08:
ADC NC should be tied to GND according to the data sheet.

hartytp, 2018-11-08:
We're driving the ADC AVDD and DVDD from the same supply. Usually, I would expect to provide separate supplies for these (e.g. connect the DVDD to P3V3A via an independent 33R resistor with a 1uF||100nF capacitor). @cjbe I guess the reason you chose not to do that in your design is that we keep the digital interface inactive during sampling so we don't need to worry about noise on DVDD. Is that correct, or is it worth isolating?

hartytp, 2018-11-08:
> I used tiny SMD shielded inductors that saturate at 3A.

Since the MAX1968 max output current is +-3A isn't it worth using inductors that saturate at something a bit higher than 3A. We don't want to operate the converter with the inductors close to saturation do we?

hartytp, 2018-11-08:
@gkasprow it's definitely worth adding some schottky diodes to protect the ADC inputs in case the user messes up the wiring harness and connects the TEC outputs to the thermistor input. That would put 5V across the ADC inputs, resulting in an ADC input current beyond the 10mA abs maximum rating. Just use the same diodes on Sampler (to be extra sure, we can split the 100R into 50R->diodes->50R->ADC).

hartytp, 2018-11-08:
not that it really matters, but afaict it's more natural to connect LX2 to tec_n and LX1 to tec_p (i.e. swap the _p and _n around on the tec connectors).

hartytp, 2018-11-08:
Is it worth having a CMC on the TEC outputs to reduce EMI (there is quite a lot of common-mode noise on those outputs)?

hartytp, 2018-11-08:
The TEC driver recommends having a 100uF electrolytic somewhere on the supply to smooth out ripples. Worth adding one?

hartytp, 2018-11-08:
> The TEC driver recommends having a 100uF electrolytic somewhere on the supply to smooth out ripples. Worth adding one?

NB on @cjbe's design this wasn't necessary since we had an external 5V supply with a big output capacitor. However on Thermostat the 5V is sourced from an internal 5V SMPS.

gkasprow, 2018-11-08:
We have to relay on 5V supply itnernal protection then.

hartytp, 2018-11-08:
> We have to relay on 5V supply itnernal protection then.

That can supply 8A, which is more than enough to trash one of the max chips.

I don't think there is any way of getting around this; mis-wiring the connector so that TEC_N, but not TEC_P, is shorted to V+ or GND will always kill this chip.

hartytp, 2018-11-08:
@gkasprow anyway, I don't think this needs to turn into a multi-round review. Do what you think is best based on my feedback, and let me know when you're happy with the design. Once that's done I'll order a few boards from TS to play with.

jordens, 2018-11-08:
Just a heads up: we will want a version of Thermostat with an STM32 at some point in the near future.

gkasprow, 2018-11-08:
@jordens I remember discussion about CPU selection and @sbourdeauducq was against STM32 + Ethernet PHY and preferred TM4C. That's why we switched to it. We already used in Booster design STM32 chip.

jordens, 2018-11-09:
I'd prefer a STM32 for the custom PWM solution for various reasons and IMHO the WizNet+MTD is a nice simple solution. TM4C/STM32+MTD is probably a lot of development investment that may not be worth it given the other approaches. I'm not too worried about the external classic PHY instead of an integrated one.

hartytp, 2018-11-09:
> Just a heads up: we will want a version of Thermostat with an STM32 at some point in the near future.

What's your plan firmware-wise? Write your own (open source/public?) or wait for Greg's student to post something?

> I'd prefer a STM32 for the custom PWM solution for various reasons 

Sounds good. Other than that, are you happy with the design discussed here?

hartytp, 2018-11-09:
For the microprocessor here, shall we pick the one that makes most sense to use on the servo card and use the same one here?

jordens, 2018-11-09:
Depending on funding and software state by then we might write our own firmware, probably open. But in the current projects this is not really urgent yet and it's still a while until we'd need to look into it.
I'm happy with the general design but haven't had time to do a detailed schematic and layout review.
Yes. using the same µC for the servo and this would be wise. We're also using the STM32 µCs with MTS/WMS modulating/demodulating servos so it's a nice fit.

hartytp, 2018-11-09:
Good, thanks for clarifying.

Let's see how Greg's student gets on then.

gkasprow, 2018-11-09:
My student wrote the code for TM4C for first version of Thermostat. It is based on FreeRTOS. It should be possible to move it to STM32. But the student needs to finish his engineering thesis soon.
If we are going to develop firmware using students, I'd stick with TM4C because some work is already done.
@jordens what is the difference in PWM implementation on TM4C and STM32? They are essentially same architecture. Is TM4C PWM so complicated?

hartytp, 2018-11-09:
@gkasprow that's fine. TM4C is a useful stop gap until we have the firmware ready for the custom solution.

I'd still like to get the new boards ready for manufacture reasonably soon if possible while this is all still fresh in my mind.

gkasprow, 2018-11-09:
If @jordens wants to play with it, he will have working solution to start with including PWMs.

gkasprow, 2018-11-10:
The diode I proposed has only 0.5pA of leakage current
It will achieve 0.4nA current at roughly 100degrees.
[image]

gkasprow, 2018-11-10:
I already added the diodes. If we discover that they ruin our temp stability, we can always remove them.

hartytp, 2018-11-10:
Sounds good @gkasprow. Anyway, am I right in thinking that the leakage will be significantly lower when the reverse bias is only 1V or so?

Once we have the boards, we can easily make a temperature stability measurement to confirm that the performance matches our calculations. Use one channel to perform an open-loop measurement of the "temperature" of a precision 10k resistor, and use the other channel to temperature control the Thermostat board. Look at the reported "temperature" of the resistor as we change the actual temperature of Thermostat.

gkasprow, 2018-11-10:
This is what I expect and probably leakage current won't be an issue here.

gkasprow, 2018-11-10:
@hartytp please have a look at newest release.
I also created dedicated repo for Thorlabs version. 
https://github.com/sinara-hw/thermostat_thorlabs

gkasprow, 2018-11-10:
These are two modules that don't have many common things. There won't be common firmware, chipset is different.

gkasprow, 2018-11-11:
CMC wont saturate up to 5A
Changed connectors to 5pin ones with GND
Grounded ADC AIN4

hartytp, 2018-11-11:
Splendid :)

No further feedback from me then. Let's give others until end of Weds to give any further comments and then produce a batch (email Greg/Me/Wojciech if you want some).

hartytp, 2018-11-11:
@gkasprow well, the only other comment I have is about the layout. The 24-bit ADC is packed in tightly between the microprocessor and the TEC driver, so there is quite a bit of potential for EMI issues.

I don't think there is much we can do to improve this without making the board quite a bit bigger, so let's build one and have a go. We can easily test this by using a precision 10k resistor instead of a thermistor and looking at the rms noise in the ADC measurements. We can also look at the change in measured temperature as a function of the TEC current with the device running open loop. 

Anyway, hopefully this won't cause any issues, but it's worth being aware of it.

gkasprow, 2018-11-11:
I designed it in such way to make sure the return currents don't pass via the ADC circuit.
I don't thin we would get any EMI issues.

hartytp, 2018-11-11:
Okay, good! :)

jordens, 2018-11-13:
* It's weird to use a PWM to control a PWM. If the MAX is just there to provide V/IP/IN limiting, some PWM-to-current loop, and an H bridge, then it seems misplaced. If there was no filter and just an H bridge driver from two suitably modulated pins of the µC and I/V sensing, that would be more straight forward. And you could do really nice sigma-delta modulation to get efficient spectral shaping of the output noise compared to that MAX. It might need about the same space. If the PID loop is digital, I would really want to stay digital all the way up to the H bridge. I am not doubting that you can get good performance with this. This is just to point out that this is an irritating design mismatch. Isn't there anything from the H bridge world that does this better?
* Anti windup wont be straight forward (for the reason above). There is either wasted headroom between the actual limit and what's reported as limit or there is a dead zone where the reported limit (in the digital PID) is reached after the driver is already railed.
* Should be possible to also connect I2C on the second TEC to the CPU. I2C2 or I2C3 (when moving STAT) seem doable on quick glance.
* Please expose the UART not on individual pads but on a pin header.

hartytp, 2018-11-13:
> It's weird to use a PWM to control a PWM. If the MAX is just there to provide V/IP/IN limiting, some PWM-to-current loop, and an H bridge, then it seems misplaced. If there was no filter and just an [...]

For me, using the MAX chip mainly comes down to convenience. Its performance is pretty good, it packs a lot into a small board area with fairly low EMI, it's relatively cheap, and, most importantly, it reduces the amount of design + software development work we need to do to get a high-quality board up and running.

If we're going to switch to a roll-your own solution that requires more work (and probably costs more, requires more board space etc) then there should be a strong performance reason to do so. Do you think the MAX IC's noise etc performance won't be good enough for us?

> This is just to point out that this is an irritating design mismatch

I agree that it's conceptually less neat than rolling our own, but that doesn't mean it's not the best means of achieving the result we want.

If you're keen on an H bridge approach, can you suggest a topology + some components you'd like to use?

jordens, 2018-11-13:
I'm not suggesting to roll your own H bridge. There are small fully integrated H bridges with gate drivers and power-MOS, some even with current sensing, and multiple channels in one package. And I am not doubting the specifications of the MAX. I'd just try to get a bridge driver that doesn't come with its own PWM modulator and current loop. Instead I would look at those integrated bridges/drivers and add the missing features (current/voltage sensing/limiting). That all looks well within the scope of the on-chip ADCs (I/V sensing and limiting is not a precision or high resolution job). Instead of one modulated control signal going to the driver you'd now have two. You'd save on the filtering components and you'd get voltage monitoring as well. What else is missing that you need the MAX for? Does it have better performance than a typical bridge driver? From the datasheet it is meant to be used in an analog loop, not in a digital one.
Bringing the digital signal all the way to the bridge instead of reconstructing it with a filter and steering a PWM to a current sensor would at least allow you to do noise shaping and get voltage sensing for free. It looks likely that it is also better suited for low thermal mass applications where the switching pattern needs to change.
Performance is probably going to be fine for most applications. And if it hinges on me finding a better bridge chip doing the design, then I'll pass ;) And this is probably fine as a first iteration.
In any case, don't you want to wait and debug the parts of the design shared with the Thorlabs version first before signing this one off?
We'd be interested in testing this one for ultra-low thermal mass control at some point as well.

hartytp, 2018-11-13:
> Performance is probably going to be fine for most applications. And if it hinges on me finding a better bridge chip doing the design, then I'll pass ;) And this is probably fine as a first iteration.

Well, it hinges on someone doing the design. I'm short on time right now and the MAX IC will be absolutely fine for all the applications I have in mind, so it won't be me who does all the legwork for this. If you can convince someone else to do the design work required for this then it's fine by me.

What do you think @gkasprow 

> In any case, don't you want to wait and debug the parts of the design shared with the Thorlabs version first before signing this one off?

By this point, there isn't much common between the two boards, and I have some temperature control needs coming up for which the Thorlabs chip is a bit marginal. So, unless someone is planning to test the Thorlabs version of Thermostat reasonably soon then I'd prefer to press on to manufacture.

gkasprow, 2018-11-13:
@jordens The MAX works at 1MHz. Even with 10 bit PWM you would need 1GHz clock in CPU. One can do some tricks with 8 bit PWM and noise shaping but IMHO it is not worth the effort.

jordens, 2018-11-13:
The PWM from the CPU to the MAX will also need dithering to get the needed resolution. There is no change from that.

jordens, 2018-11-13:
Even if you were to stay at plain PWM from the CPU, you get even more resolution by adjusting the frequency. The continued fraction algorithm to do that will fit just fine into the ~10ms sample rate. I assume that's what @cjbe was already doing.

hartytp, 2018-11-13:
> Even if you were to stay at plain PWM from the CPU, you get even more resolution by adjusting the frequency. The continued fraction algorithm to do that will fit just fine into the ~10ms sample rate. I assume that's what @cjbe was already doing.

@cjbe

gkasprow, 2018-11-13:
I"d like to treat the version with MAX chip as a first iteration.
 Doing own HW bridge with cycle per cycle current sensing needs quite sophisticated mechanisms inside of the CPU.
Once I did laboratory grade BLDC motor driver using brand new special AVR chip dedicated for that tasks with special high resolution PWMs with HW dithering, synchronous ADCs, HW protections, etc but it took  enormous amount of time due to bugs in the silicon. Finally I had to build required functions writing critical parts in assembly. And I burned a lot of MOSFETs during testing because HW protection functions (programmed using fuses) did not work as expected causing spectacular failures by switching top and bottom switch at the time :)
Every design can be made better, but if MAX chip does the job and simplifies R&D, let's use it. If we discover that it does not meet our expectations, we will switch to something else.

hartytp, 2018-11-14:
@gkasprow It sounds like the plan is to keep the MAX IC for Thermostat v2.0. If we decide it's not good enough, we can replace it in v3.0. Thermostat v2.0 will give us a high-performance board that we can develop most of the required firmware on.

Please could you update following @jordens comments (copied below) and then upload a release and close this issue?

- [ ] also connect I2C on the second TEC to the CPU. I2C2 or I2C3 (when moving STAT) seem doable on quick glance.
- [ ] Please expose the UART not on individual pads but on a pin header.

gkasprow, 2018-11-17:
done for both designs.

hartytp, 2018-11-17:
Thanks!

Okay, let's close this. We'll keep the new TEC driver design in mind for v3.0

closed 2018-11-17 by hartytp