sinara-hw/Thermostat #6: UART pins

hartytp, 2019-01-29:
@gkasprow right now the USB UART is only exposed via test points. Can you add a pin-header for the next version so we can use it with something like https://www.amazon.co.uk/UART-CP2102-Module-Serial-Converter/dp/B00AFRXKFU

hartytp, 2019-01-29:
also needs power and ground

gkasprow, 2019-01-29:
the Thermostat has a pin header so you can use it without a problem.
[image]

gkasprow, 2019-01-29:
btw, a box with assembled Thermostat is laying since a few days on my desk. I didn't have time to play with them...

hartytp, 2019-01-29:
@that look different to the rendering in the pdf release. Is the pdf out of date?

hartytp, 2019-01-29:
Cool are those the boards we ordered? If so can you ship them?

closed 2019-01-29 by hartytp

gkasprow, 2019-01-29:
Yes, I planned to make burn tests only.

hartytp, 2019-01-29:
Ok cool. Well looking forward to getting my hands on them :)

hartytp, 2019-01-29:
Please do make sure the release is up to date though

gkasprow, 2019-01-29:
Oh, I've just noticed that this change was added after the boards were shipped for production.
 So will be available in next release.

hartytp, 2019-01-29:
Makes sesne