sinara-hw/Thermostat #8: Prototype ADC AFE

hartytp, 2019-02-01:
- [ ] I'll buy an ADC eval board and dead-bug the AFE components from Thermostat v1.0 onto it
- [ ] use 10k precision/fixed value resistors to mimic the thermistors. connect them by ~5m of screened twisted pair to provide a realistic antenna for pickup
- [ ] play with eval board and post optimized registers here
- [ ] post a log of the equivalent temperature of the resistor and see how good we expect the stability to be
- [ ] later on, repeat the same measurement with Thermostat. Look for things like changes in the measured temperature with TEC current.

gkasprow, 2019-06-21:
@hartytp are U sure we want to use ADC from Stabilizer? It has only 16 bits.

gkasprow, 2019-06-23:
not relevant anymore

closed 2019-06-23 by gkasprow