sinara-hw/Thermostat #12: replace microprocessor with STM32H743ZIT6

hartytp, 2019-02-21:
to match stabilizer.

Please attempt to hook everything up (ADCs, peripherals, etc) to maximize the overlap between Thermostat and Stabilizer.

gkasprow, 2019-06-21:
@hartytp are U sure we want to use ADC from Stabilizer? It has only 16 bits.

hartytp, 2019-06-21:
CPU ADC is just like there for diagnostics eg TEC driver zero current measurement. We’ll keep the same 24 bit ADC for the temp measurement

jordens, 2019-06-21:
Going for the H7 stuff in Thermostat is not necessarily the right thing to do:

* The chips seem a bit immature. E.g. they are currently going through a major Silicon revision that has significant impact on the functionality, the register maps, the features and the performance. This is not something you want to deal with.
* There may be availability issues. There have been in the past.
* Nothing from the added performance is really needed.

gkasprow, 2019-06-21:
So for which chip we go? There should be some pin compatible ones.

gkasprow, 2019-06-21:
What about using same as in the  Booster?

pt., 21 cze 2019, 12:46 użytkownik Grzegorz Kasprowicz <kasprowg@gmail.com>
napisał:

> So for which chip we go? There should be some pin compatible ones.
>
>

jordens, 2019-06-22:
Yeah. Maybe consider something like the STM32F427ZI. That's the same family as the F4 in Booster, a bit bigger, more peripherals, more flash, more ram, a bit faster. It has a pin compatible path upwards all the way through the F7 and H7 lines.

gkasprow, 2019-06-22:
I need 4 SPI because one is blocked by JTAG/SWD
I used STM32F745VGT6 in QSP100 package. I don't have enough place for 144 one.
If you accept it, I will publish schematics. I finished them during the flight from SF to WAW yesterday.

jordens, 2019-06-22:
The LQFP100 (I assume that's what you mean) pinout is a bit different from between the F4 and F7 series in the [upgrade path](https://www.st.com/resource/en/application_note/dm00337702.pdf). Between [F7 and H7 it's fine](https://www.st.com/resource/en/application_note/dm00164538.pdf). F7 currently doesn't seem to have a terribly active Rust community either. If the choice is F7 or H7 then maybe going for H7 is better. but make sure it's silicon rev V and the part is common and widely available (in dev kits, Nucleo or Discovery boards).

gkasprow, 2019-06-23:
OK, I can fit LQFP144. So I will use STM32F427ZI.

closed 2019-06-23 by gkasprow