sinara-hw/Thermostat #13: add PDF for v1.0

jordens, 2019-03-13:
@gkasprow the only PDF of the current version I can find is https://github.com/sinara-hw/Thermostat/issues/4#issuecomment-436790828
Could you add one to the v1.0 release?

gkasprow, 2019-03-13:
done.

jordens, 2019-03-14:
I can't find it. And the v1.0 tag is not published as a release.

hartytp, 2019-06-24:
@gkasprow is this done?

jordens, 2019-06-24:
No.

gkasprow, 2019-06-25:
The pdf is here
https://github.com/sinara-hw/Thermostat/releases/tag/v1.0rc1

jordens, 2019-06-25:
That's not v1.0 and doesn't correspond to the v1.0 board I got at all.

hartytp, 2019-06-25:
Yes it’s an dc. I don’t think there was ever a pdf for the production version

hartytp, 2019-07-11:
@gkasprow please can we have a PDF of v1.0? The one in releases is still an RC and not the production version.

gkasprow, 2019-07-13:
OK, I've just noticed that pdf was generated in the wrong location.

closed 2019-07-13 by gkasprow