sinara-hw/Thermostat #16: enclosure mechanics

jordens, 2019-03-15:
* The tantalum cap in the corners next to the green terminals and the IDC connectors are too close to the enclosure
* There are PCB traces at the edge that risk being scratched and shorted by the enclosure when inserting
* The pins from the RJ45 and the green terminals touch the enclosure.
* Inserting it in the second slot from the bottom doesn't work because of the RJ45 connector.

I'd make sure that there is enough clearance to the side slots (tantalum, IDC, other parts parts and traces) and that the pins are cut.

gkasprow, 2019-06-23:
we are switching to 4 layer PCB anyway so this will solve some of the issues.
THT pins need trimming before placing in the enclosure.

hartytp, 2019-07-15:
I also had issues with the PoE module colliding with the side on one board. Maybe it was soldered on at a bit on an angle, but I had to bend it in to prevent some thicker components from colliding with the enclosure.

gkasprow, 2019-10-26:
All issues fixed. At least on the 3D model.

closed 2019-10-26 by gkasprow