sinara-hw/Thermostat #18: add GPIO header?

sbourdeauducq, 2019-05-14:
What about adding a GPIO header connected to the otherwise unused pins of the STM32?

The immediate application I have for it is connecting a current source driving a laser diode, to study its behavior under different temperature and current conditions. With that GPIO header, the diode driver could be a simple stripboard add-on.

gkasprow, 2019-05-14:
What about using same connector as for Stabilizer , routed to same CPU
pins, so one can reuse same mezzanines?

hartytp, 2019-05-14:
The Stabilizer connector is probably too big for Thermostat. I'd prefer not to increase the board area too much for a niche feature.

jordens, 2019-05-14:
I would prototype a current driver on Stabilizer or any of the other µC (e.g. Nucleo) boards and use Thermostat separately.
If there is space for a connector without disturbing the layout, introducing clearance problems, or changing the board size I'm ok with adding one.

sbourdeauducq, 2019-05-14:
> I'd prefer not to increase the board area too much for a niche feature.

Niche?
I thought one of the main applications of the thermostat was to stabilize the temperature of laser diodes, and those almost always needed programmable current sources.

gkasprow, 2019-06-23:
what signals would you want on such header?
I have some board are and can place 10 pin 2.54mm header.

gkasprow, 2019-06-23:
I can route 2 ADC, 2 DAC and some GPIO

sbourdeauducq, 2019-06-23:
Okay, sounds good!

closed 2019-06-23 by gkasprow