sinara-hw/Thermostat #19: AD590 temperature sensor support?

sbourdeauducq, 2019-06-18:
The AD590 seems to be a popular option in several commercial or academic ECDL designs, and it might be interesting to support it.

gkasprow, 2019-06-19:
it is current output sensor, with 1uA/K conversion ratio.  It gives 300uA at room temperature and 220uA at -55degC.  The only issue is the power supply. It needs a minimum of 4V. We would need to hook it to 5V rail and measure the voltage on a resistor connected to GND. If we use existing 5k11 resistor, we would get
1.55 V at room temperature so 5V supply would be not enough. So the resistor must be lowered to roughly 3k to maintain the best SNR. The negative terminal of the ADC would need to be shorted to GND. 
So, there is a way to do it but it would require additional LDO that stabilizes 5V to let's say 4.8V, some jumper that chooses the lower value of the resistor and yet another one that connects LDO output to positive ADC pin.

hartytp, 2019-06-19:
The current thermostat design is optimized for ratiometric measurements of resistances ~10k. In particular, this allows us to avoid using a high-spec reference, since the reference voltage/noise common-modes out.

The AD590 is quite different since we'd need to measure an absolute current/voltage across a sense resistor. 1mK/300K=3ppm. So the noise/drift on the reference has to be very low to achieve decent stability. Moreover, to get 0.5C accuracy on 300C you need the reference to be good to 0.1%, which the current implementation isn't.

So to make this work we'd need to overhaul the ADC supply/reference.

hartytp, 2019-06-19:
Assuming we used a 1V sense resistor, the 1mK=1nA*1kOhm=1uV. So, we'd need to be very careful to make a proper differential/kelvin-connection measurement across the sense resistor. The current circuit topology doesn't allow that. I suspect that just grounding one ADC input and making a single-ended measurement of the sense resistor voltage won't give very good results.

hartytp, 2019-06-19:
In other words, this is something we can do with a handful of jumpers (maybe FETs, but that would need some thought) and an overhaul of the ADC reference/supply, but it's not a trivial change.

I wouldn't support going down this rout unless someone wants to take ownership of it and provide a sketch of what conenctions we'd need, along with a noise/stability analysis and a specification of what they're trying to achieve. If no one has time to do that right now then let's leave it to a future design revision (or maybe a design variant).

jordens, 2019-06-19:
Thermistors are also about 20 times more sensitive than a PTAT. That translates to controller performance accordingly.

gkasprow, 2019-06-23:
nothing to do here, closing.
I don't have space on PCB to add jumpers.

closed 2019-06-23 by gkasprow

hartytp, 2019-06-24:
I think the best thing is to make a separate PCB for the AD590 if people need that. The front-end circuit required for the AD590 is sufficiently different to RTDs to make supporting both tricky without significant space/complexity increases.