sinara-hw/Thermostat #24: ic1 sw pins connection missing

hartytp, 2019-06-25:
why aren't pins 10 and 48 tied together?

hartytp, 2019-06-25:
@gkasprow answering my own question, I guess the reason is that there are two banks of SW pins on opposite sides of the IC and you don't want to route them together.

Also, I haven't checked the compensation for this reg, but assume you're happy with it?

gkasprow, 2019-06-26:
I took the values from the datasheet.

hartytp, 2019-06-26:
@gkasprow looking at the example layout (fig 6-1) the two sets of switch pins should be tied together with a thick trace.

hartytp, 2019-06-26:
@gkasprow I didn't see much guidance in the data sheet on the choice of Cinj and Cff. It looks a bit like they have to be chosen empirically (depends a lot on the ESR of the capacitors used). Are you happy with the values there? Did you check the stability/noise of this regulator?

gkasprow, 2019-08-18:
I took them from the [reference design](https://www.arrow.com/en/reference-designs/mic45116-1ymp-ev-evaluation-board-for-the-mic45116-1-20v6a-dc-to-dc-power-module/98683d59703926eaa41bfb0839684ba1)

gkasprow, 2019-10-26:
I think I have to look at it once again and verify its stability with a step response test.
The sudden death of the 3V3 converter made me suspicious. I'm using this configuration in a few other designs and nobody complained so far but since a lot depends on load, who knows...

gkasprow, 2019-10-26:
I can also do it with V2.0 since I don't have any thermostat in my hands.

hartytp, 2019-10-27:
Happy to send you one if it would help?

gkasprow, 2019-10-27:
I have other design with an identical converter so can use it for testing. 
I do a very simple test:
 Connect a scope and a pushbutton with a 2R/5W resistor in series.
One probe goes to the switch, another to the power rail. Then trigger on falling and rising edge and observe the response. It cannot oscillate. If it does, there are stability issues.

hartytp, 2019-10-27:
Good idea. FWIW I think we’re running at a quite light load, not sure how that impacts stability

hartytp, 2019-11-30:
@gkasprow are you happy the layout for this ic follows the reference design?

gkasprow, 2019-11-30:
yes.

closed 2019-11-30 by hartytp