sinara-hw/Thermostat #26: DAC PWM

hartytp, 2019-06-26:
@jordens did you want a FET that can be used to PWM the DAC output?

jordens, 2019-06-26:
That would be useful. Yes. Before the rc filter (can be DNP if it's not needed for the dac) from v1.0

hartytp, 2019-06-26:
I'm fine with that so long as someone thinks through the usual things like Ron etc to make sure that it won't degrade the performance when not used.

hartytp, 2019-06-26:
NB the DAC is really a 16-bit DAC with an internal modulator/interpolator. So we're already doing a scheme like that at a very basic level...

gkasprow, 2019-06-26:
just FET or analog switch?
FETs have high charge injection unless you take one of [these](http://www.calogic.net/pdf/SST211_Datasheet_Rev_A.pdf)

hartytp, 2019-06-26:
ill Leave that decision to @jordens fwiw though I don’t think charge injection is an issue due to the 2 pole rc filter (we’re doing pwm so need a good filter)

I assume well so something like have a 1k resistor after the dac and then a shunt fet.

To check though, the charge injection is only an issue when the fet switches right, not when it’s static?

jordens, 2019-06-26:
With that 18 bit DAC and 1 LSB DNL it's fine. Plenty of resolution even for small loads. No need for the FET to PWM-multiply the DAC output.

jordens, 2019-06-26:
Really interesting DAC by the way. Nice for high resolution low noise actuators inside slowish control loops.

closed 2019-06-26 by jordens

gkasprow, 2019-06-26:
Yes, Moreover charge injection is usually symmetrical so the avarage charge is 0.