sinara-hw/Thermostat #27: DAC reference

hartytp, 2019-06-26:
@gkasprow shouldn't we use a 3V reference for the DAC to match it to the TEC driver? Currently we're throwing away nearly an entire bit of resolution by using the wrong reference.

gkasprow, 2019-06-26:
True. We can also add resistive divider but the reference is probably cheaper.

hartytp, 2019-06-26:
Nb the dac still needs a decent 5V rail as it doesn’t support 3V vcv

hartytp, 2019-06-26:
> True. We can also add resistive divider but the reference is probably cheaper.

Actually, we can probably get away with a divider made from 0.1% resistors. After all, the DAC offset is 10mV, which is 0.3%, so a small error on the reference won't harm us.

hartytp, 2019-06-26:
And, DAC gain accuracy is only 1.5% so no point having an ultra-precise reference.

The main worry I'd have would be the higher reference impedance degrading the DNL. @gkasprow am I right in thinking that for this kind of resistor string DAC, that shouldn't be a problem since the ref current is constant? Or do we need a proper buffered reference here?

gkasprow, 2019-07-12:
@hartytp AD5680 has already buffered output. So we can add resistors.

hartytp, 2019-07-12:
Won’t it be an issue that the ref impedance at the ac will be high due to the resistors? Doesn’t the dac need a low impedance red?

gkasprow, 2019-07-13:
Oh, I was thinking about adding voltage divider at the DAC output to match the range.

gkasprow, 2019-07-13:
we can also use the ADC 3.3V supply as a reference.

hartytp, 2019-07-13:
Aah ok. So long as the tec and cpu ADC inputs are high z, a divider on the dac output sounds like the best option

gkasprow, 2019-07-13:
MAX1968 has 1MOhm input resistance. We don't really care about DC accuracy. So resistive divider in order of 1k should be fine.

hartytp, 2019-07-13:
Sounds good

closed 2019-08-18 by gkasprow