sinara-hw/Thermostat #28: mesaure DAC output on microprocessor

hartytp, 2019-06-26:
@sbourdeauducq should we wire the DAC output to a microprocessor ADC input? The DAC only has 12-bit accuracy (10mV offset), which is comparable to the TEC driver. No point calibrating the TEC if we don't calibrate the DAC as well.

hartytp, 2019-06-26:
This also means we're not sensitive to offsets in the microprocessor ADC since they common-mode out on the two measurements (we're effectively finding the DAC word that makes the DAC output equal to the TEC ref value).

gkasprow, 2019-06-26:
>  The DAC only has 12-bit accuracy (10mV offset)

Do you mean the CPU ADC?

hartytp, 2019-06-26:
No the 18 bit DAC is only 12 bit precise as it’s designed for in loop applications like this. 18 bit accurate dacs cost $$$ this one is cheap. Stability and noise are good so it’s fine for our application

hartytp, 2019-06-26:
CPU ADC offsets don’t matter so long as we measure the dac outputs and tec ref on the same ADC

gkasprow, 2019-06-26:
I added another RC to isolate the DAC from the CPU ADC sampling current pulses

hartytp, 2019-06-26:
When did you add that?

gkasprow, 2019-06-26:
a few seconds ago.
[image]

hartytp, 2019-06-26:
Did it make any difference?

hartytp, 2019-06-26:
In the first instance, let's try to focus on allowing you to reproduce my results on an unmodified board. See if you can get +10V on channel 0 and -10V on channel 1 by flashing according to Robert's instructions. If not, I'll have another look at the FW I sent you in the morning.

gkasprow, 2019-06-26:
It's thermostat issue, not stabilizer :D

hartytp, 2019-06-26:
:) sorry. Doing too many things at once

hartytp, 2019-06-26:
Looks good

jordens, 2019-06-30:
When using the DAC and PWM Vref calibration values that you would obtain from this, I'd be very careful what to do with them. Initial one-time offset calibration, monitoring and correcting the limit PWMs later on are fine. But I don't think repeated feedback of those values onto the Iset DAC output are a good idea.

hartytp, 2019-06-30:
@jordens yes I’d go for a power on cal and leave it there.

closed 2019-08-18 by gkasprow