sinara-hw/Thermostat #29: schematic review

hartytp, 2019-06-26:
I've checked the supplies and analog parts and am happy. @sbourdeauducq / @jordens would you mind checking that you're happy with the digital side of things?

gkasprow, 2019-06-26:
I posted also CubeMX config files, so one can with one click generate all init software for specified pin assignment and peripherals.

jordens, 2019-06-30:
I would recommend testing and characterizing the analog parts sufficiently before doing the redesign.

hartytp, 2019-06-30:
@gkasprow do you have a thermostat set up with firmware already? Do you think you could have a quick look at the temperature stability this week? Just connect a low temp-co 10k resistor in place of a RTD and log the temperature for a few hours.

gkasprow, 2019-06-30:
I don't have any Thermostat at the moment.

hartytp, 2019-06-30:
I have a spare one I’m happy to send you (btw did TS send you a stabilizer current sense AFE?)

gkasprow, 2019-06-30:
I gave one to Creotech, can take it back if they don't use it. I didn't get stabilizer AFE from TS. Note that at the end of next week I'm leaving for the holidays. I will stay for one month in the US.

hartytp, 2019-06-30:
> I gave one to Creotech, can take it back if they don't use it.

Okay, I'm happy to overnight you one tomorrow if it would help.

> Note that at the end of next week I'm leaving for the holidays. I will stay for one month in the US.

I think it's important that we try to get the boards off before you leave. I need to get hw produced before too long so that M-Labs can write the new firmware. Let's do as much testing as we can (I will try to do some as well), but still aim to get production files ready by the end of the week. (@jordens / @sbourdeauducq can you try to finish your design reviews in the next few days?)

> I didn't get stabilizer AFE from TS.

Odd, I asked them to send you one. They must have forgotten. I can ship at the same time as Thermostat or bring one to NACTI if that helps?

gkasprow, 2019-06-30:
You can bring to NACTI.

hartytp, 2019-06-30:
Okay. I’ll do what testing I can this week (anything anyone else can do gratefully received) but let’s aim to get the next design rev ready for production before @gkasprows holiday

hartytp, 2019-07-03:
@jordens @sbourdeauducq other than the thermal management have you completed your design review?

jordens, 2019-07-03:
I have reviewed everything in my area of expertise except the µC pin assignment. With the pending changes, that'll change quite a bit. Someone should take a good long look at that the choice of peripherals, timer channels, w.r.t. pins, internal buses, dma architecture, peripheral capabilities etc. I won't have much time to do that in the next weeks.
And I would still recommend that someone does a decent amount of characterization and testing of the current device. Both ADC and TEC driver.

gkasprow, 2019-07-03:
For this purpose, I use STM32CubeMX which does most of the job you mentioned.

jordens, 2019-07-03:
Sure. Some. It didn't cover the peripheral specific properties for stabilizer for example.

closed 2019-10-26 by hartytp