sinara-hw/Thermostat #30: PoE AT detection

jordens, 2019-06-29:
For compliant operation the device can't pull more than 15.4 W unless we detect that we are allowed to. We are several watts above that limit and during the typical turn on transients when the thermal masses are cooled/heated from room temperature that limit will typically be exceeded.
The behavior if we don't detect it properly is guaranteed to be confusing (PSE trip).
What about putting the optocoupler and the AT detect pin to the CPU?

gkasprow, 2019-06-29:
Good idea. We already did that for another [project.](https://github.com/BerkeleyLab/Marble/releases)

gkasprow, 2019-06-29:
shall we do the same for Stabilizer? With mezzanines, we may exceed 15.4W

jordens, 2019-07-01:
Yes same for stabilizer

closed 2019-07-12 by gkasprow