sinara-hw/Thermostat #31: Ad5300 ripple capacitor

jordens, 2019-06-29:
Should be 47 uF, is 460.
Imo going to 3*220 would be nice.

Also and maybe more importantly:
Is there some way to mitigate the ground ripple?
Since it's an actual isolated transformer I would not have expected ground loops that bad.

gkasprow, 2019-06-29:
it does not make any difference because tolerance of electrolytic caps is usually 20% or worse.
A higher value may cause PSE tripping.

jordens, 2019-06-30:
What about adding the bypass caps (4.7nF 2000V) around the module and the choke for emi filtering as in the datasheet (fig13)?

gkasprow, 2019-06-30:
Good idea. The same with Stabilizer.

jordens, 2019-06-30:
Although the RJ45 connector seems to contain some caps already.

gkasprow, 2019-06-30:
Yes, but they are in series with 75Ohm.
The CMC and cap mentioned in the datasheet are used to make a bypass for PoE module Flyback converter.
The converter injects noise between primary and secondary side. Cap placed between them shorts the interferences and CMC is used to break the ground loop.

jordens, 2019-06-30:
But interestingly there are no caps on the non-data pairs.

gkasprow, 2019-06-30:
The RC network inside of the RJ connector serves a different purpose. It is so called "Bob Smith" termination. However, there are [doubts](https://www.eetimes.com/document.asp?doc_id=1277940#) if the values are correct.

gkasprow, 2019-06-30:
The conclusion is that we would need CMC to keep Bob Smith termination working. This affects EMI performance.

hartytp, 2019-06-30:
> Since it's an actual isolated transformer I would not have expected ground loops that bad.

To be clear, are you referring to https://github.com/sinara-hw/Stabilizer/issues/17#issuecomment-506049460

hartytp, 2019-06-30:
> The CMC and cap mentioned in the datasheet are used to make a bypass for PoE module Flyback converter.

@gkasprow which CMC? I only see ferrets there

hartytp, 2019-06-30:
@gkasprow can you sketch out your recommended circuit for this?

hartytp, 2019-06-30:
When I was looking at noise on Stabilizer, I didn't see a difference between powering from PoE or an external power supply. However, I did notice that when I turned the the PoE switch on with no cables connected (other than the power supply) I suddenly saw a lot of switching spikes on my scope. Maybe it's an issue with the particular switch I'm using, but maybe we should be prepared for some grounding/noise issues with PoE...

jordens, 2019-06-30:
Yes. It looks very different depending on the switch, the PoE mode.

gkasprow, 2019-07-12:
added CMS and caps.

closed 2019-07-12 by gkasprow

hartytp, 2019-07-12:
Nice. Can I see the circuit you added? Did you use cmcs or inductors (as in the data sheet?)

gkasprow, 2019-07-12:
[image]

hartytp, 2019-07-12:
Wouldn’t inductors/fbs be better? We want to block both the cm and the dm signal, but the choke only gets the cm.

hartytp, 2019-07-12:
Also do c3 and c12 need to be he caps? And should we use 4.7nF input to output caps instead as in the data sheet?

gkasprow, 2019-07-13:
True, we can add inductors, but CMCs are much smaller. The main issue is common-mode noise caused by the DC/DC converter. The inductors in the datasheet of the PoE module are ferrite beads that have Z close to zero for SMPS frequency. C3 and C12 need to be high voltage caps to be compatible with Ethernet isolation barrier. Their role is to make a bypass for PoE noise source.

hartytp, 2019-07-13:
Okay so long as you’re not worried about dm noise currents radiating and causing emi issues that sounds good.