sinara-hw/Thermostat #38: evaluate thermal design and cooling requirements

jordens, 2019-07-01:
From a rough calculation where the three switchers (PoE, the 12V-5V, and the MAX) are each about/at most 90% efficient, we'll be loosing more than 30% in the device (plus µC, MAC, ADC, DACs, LDO, references). At 3A*4.3V*2 that's 26W to the Peltiers and we will be burning another 9W on top on this tiny board. That number is typically reached each time during turn on for a significant duration. (I have seen it use about 17W from PoE+ when driving a single channel. That wasn't even max power.)
This means:

* In unconstrained applications we'll max out the PoE allowance (25.5W available at PD since you need to account for cable losses). This will require careful and active management in software. It's not a killer. But it means that it will be impossible to get full power out of both peltiers at the same time for a significant duration. If it's not managed in software this will be annoying and confusing at the very least.
* Since there is no way for the software to distinguish PoE (where AT_DET is deasserted) from external 12V (where it is deasserted as well) using the 12V external will mean running at 13W max from the barrel connector. If this is undesired we need a way to distinguish the power sources.
* For PoE power at full output we'd burn more than 8W on the device. For external at full beans 12 V it's more than 5.5W. This will likely overheat if continuous. One way to tackle this may be be to try to get some conductive cooling working to the Hammond enclosure: add tight coupling between the PCB layers, add exposed copper/gold strips on the outside where the PCB slots into the enclosure and design two wedges that are inserted from rear and front on top of the PCB between the enclosure and the exposed strip to make thermal contact. Admittedly that's a hack and tricky. If that doesn't work, we need to look at something else. E.g. actively limit the Peltier voltage depending on internal (ADC, µC) temperatures.

hartytp, 2019-07-01:
@jordens all good points. Thanks for the detailed work on this.

> Since there is no way for the software to distinguish PoE (where AT_DET is deasserted) from external 12V (where it is deasserted as well) using the 12V external will mean running at 13W max from the barrel connector

The barrel connectors have a connector inserted signal don't they? We should connect it to the uC.

> One way to tackle this may be to try to get some conductive cooling working to the Hammond enclosure: add tight coupling between the PCB layers, add exposed copper/gold strips on the outside where the PCB slots into the enclosure and design two wedges that are inserted from rear and front on top of the PCB between the enclosure and the exposed strip to make thermal contact. Admittedly that's a hack and tricky. If that doesn't work, we need to look at something else. E.g. actively limit the Peltier voltage depending on internal (ADC, µC) temperatures.

Yes. This shouldn't be an issue if Thermostat is used in a rack with some forced air cooling, but will be an issue in those small enclosures. @gkasprow what's your plan here?

Is it worth putting a thermometer IC that can shutdown the 5V rail if the temperature near the TEC driver exceeded a safe limit? Or do we want to do that in SW only?

jordens, 2019-07-01:
Doesn't matter if it's an issue only for some configurations. If it's an issue for one of the target configuration it should to be addressed. For the 3U version you will also need quite some flow to pull away 10 W from that small surface. Thermostat has about 3 times the power (loss) density than Urukul/Mirny. And conducted heat transfer from the PoE module in the enclosure will likely be pretty bad.
The PoE module will shut down at 70C which I think we will reach rather quickly. For safety (in the case of PoE only) that seems good to me. I would recommend optimized cooling and a proper software solution for the remaining factors.

hartytp, 2019-07-01:
> Doesn't matter if it's an issue only for some configurations. If it's an issue for one of the target configuration it should to be addressed. 
> ...
> I would recommend optimized cooling and a proper software solution for the remaining factors.

Agreed. 

I'd be interested to hear @gkasprow's feedback on our options for improving the thermal design.

gkasprow, 2019-07-01:
I would add LM75 that would shut down the power after exceeding 80 degrees.
It's not a big problem to add a circuit that makes CPU aware of the power source.
When I designed the Thermostat, the assumption was that we would work with much lower TEC power. Even 15W PoE seems to be sufficient, I used 30W one to have some margin. The idea was to keep the power output of the Thorlabs module at +/- 1.5A level.

gkasprow, 2019-07-01:
To improve heat transfer, we can do one of the two things:
- reserve some exposed copper area on the PCB where flexible copper-silicone pads can be glued which transfer the heat to the enclosure. We cannot really rely on edge heat transfer, we would need to use complicated wedge-locks.
- machine dedicated aluminum pads that would be slid between PCB and the enclosure and screwed from the bottom side to keep low thermal resistance.

hartytp, 2019-07-01:
The other thing that might be possible would be a right-angled aluminium bracket that mounts to the edge of the PCB and screws to a hole on the side of the enclosure to transfer heat. Or something like that.

hartytp, 2019-07-01:
Maybe better to make the front panel out of metal try to ensure a good thermal contact between it and the PCB. Rely on the PCB copper to spread heat

jordens, 2019-07-01:
Imo active temperature measurement and power and temperature limiting in the CPU is good. Just needs to be done properly.
Together with best effort cooling (conductive and sinking) it's just fine.

gkasprow, 2019-07-02:
That's why I like these LM75 chips. They are chip and provide thermal shutdown without any programming. So, even when somebody forgets to initialize thermal management, the board will still survive.

jordens, 2019-07-02:
In this case they wouldn't matter. The Poe module will shut off 10K earlier unless you configure the lm75.

gkasprow, 2019-07-03:
True, but this would work with DC power.

hartytp, 2019-07-03:
Imho it’s worth having a lm75 or similar to shut off the 5V and protect the circuit in case of damage (the PoE module is not in good thermal contact with the PCB so could be quite a bit cooler than the TEC driver).

Even if we dnp/remove if once the fw is working ime it’s good to have hardware fail safes during debugging/development

hartytp, 2019-07-03:
@gkasprow how hard/expensive/practical  would it be to add a heat sink + fan combo to thermostat and add a grill in the enclosure?

hartytp, 2019-07-03:
It’s nice to have the option of running close to max power for a prolonged period if we can, rather than just hitting a second/hw thermal shutdown.

jordens, 2019-07-03:
The other problem with hardware solutions is that they are much harder to debug and lead to intermittent issues. The device would cycle which might go unnoticed for a while. If there's a software solution, it would actively limit the current and still work just fine where the hardware solution wouldn't. The software solution would also signal/log/indicate the issue and the margin.

jordens, 2019-07-03:
But I'm not against the LM75. Add it.

gkasprow, 2019-07-03:
@hartytp we have some space for a tiny fan. Of course, the enclosure needs machining and some mechanical work. What about making the Thermostat twice bigger which would have full 3U with and still fit into one of the Hammond enclosures? This would make thermal management easier.

hartytp, 2019-07-03:
I think that before we take this conversation further we need a thermal model. That way we know eg how much airflow is required? Can you do a quick sim @glasprow?

hartytp, 2019-07-03:
@gkasprow can we clip a fan onto the back panel so that air flows between the two panels? That way we don’t need any extra machining. Depends on how much airflow we need since will have to be a small fan..,

hartytp, 2019-09-05:
@gkasprow what are your thoughts on this?

gkasprow, 2019-09-05:
I will make a thermal model and simulate it in various conditions. Then we will decide what to do.

hartytp, 2019-10-26:
This is the only major worry i have for this design. Not sure what we can do to stop it overheating in hose enclosures...

hartytp, 2019-10-27:
@gkasprow did you have any thoughts about this?

I'll try to finish testing v1.0 this week, so it would be good to aim to send v2.0 to manufacture next week.

gkasprow, 2019-10-27:
Once I get access to the license server, I will do some simulations. Then we will decide what to do.
But I think we will have to use forced airflow or conduction cooling in the stand-alone version. We can make a dedicated enclosure for that purpose.

hartytp, 2019-10-27:
I think that’s right: either software limit the tec power to something low or design a different enclosure with forced air cooling (use a eurocard rack I guess)

hartytp, 2019-11-18:
Trying to summarise the options that have been presented here for improving the thermal management...

NB We don't necessarily need to implement all of these, they are just suggestions.

**1. conductive cooling** (heat sinking to the case)

- now we're using a 4-layer PCB we have more ground copper, which should significantly improve heat conduction through the board
- make sure to add plenty of thermal vias near components that generate heat
- have some exposed copper around the edge of the pcb
- use some thermally conductive filler material to ensure a good contact between the board and the case (see https://github.com/sinara-hw/Thermostat/issues/59#issuecomment-553641546) . @gkasprow should we add this to the BOM with an explanatory note for the next design revision?
- the (electrically insulating) fillter material can potentially cover the whole top-side of the PCB to improve conduction
- NB the PoE module does get pretty hot (see here https://github.com/sinara-hw/Thermostat/issues/59#issuecomment-553382294 we also have other photos that give a direct measurement of the PoE module that we can post) and will not be directly helped by this

**2. I2C Thermometer**

- add an LM75 I2C thermometer. I know there is one in the ADC, but using that costs us loop BW, so it's nice to at least have the option of using a separate one for slow temperature measurements.
- hook up the over temperature alarm to the microprocessor on a pin that can generate an interrupt. 
- Also hook this up to the TEC driver shutdown pins, so that it is ORed with the microprocessor outputs (i.e. either the microprocessor or the LM75 can shut the TEC driver down).
- also hook up to an over temp LED on the FP (again, maybe OR that with a microprocessor output?)
- NB the TEC controller, PoE module and 5V SMPS all have thermal limiting, so the LM75 thermal shutdown should not be required to prevent damage

**3. software**

For a future design revision we should consider adding thermal management to the software. e.g. limiting the max power through both TECs to keep the board acceptably cool and avoid overloading the PoE module (we should also look at the AT-DET for this). Right now, one can turn both TEC drivers to max with a 1Ohm load, resulting in the PoE switch cutting the supply off. 

**4. add heat sinks**

e.g. add a pad for a low-profile SMT heat sink under the ADC. This will be useful e.g. then thermostat is used in a case with forced-air cooling

**5. fan header**

Do we want to add a header for a fan? This might be useful in stand alone applications in vented boxes.

gkasprow, 2019-11-21:
I added a simple R + Diode solution to disable both TECs. FP LED is also connected. CPU is connected to the LM75 alarm output.
[image]

I added the pad under ADC with vias and silkscreen removed
I'm not sure if exposed copper on board edges would improve heat transfer. The contact area is very low. Such exposed copper would short with enclosure and create new ground loops.
I managed to place standard fan connector but will make it DNP by default
[image]

jordens, 2019-11-21:
Sounds good.

hartytp, 2019-11-22:
will that fan header collide with the Box/rack mounting rails?

hartytp, 2019-11-22:
What's the plan for the thermally conducting pads to transfer heat to the enclosure?

hartytp, 2019-11-22:
Ok, it seems like we're nearly there. @gkasprow do you think it's feasible to aim to send Thermostat v2.0 to production on Monday?

gkasprow, 2019-11-29:
the header fits, at least according to the 3D model.
I will try the 3M conducting sheets I have in the lab.

hartytp, 2019-11-29:
Cool. 

> What's the plan for the thermally conducting pads to transfer heat to the enclosure?

?

hartytp, 2019-11-29:
> I will try the 3M conducting sheets I have in the lab.

Okay, sorry, I missed that. Will you do this before sending to production?

gkasprow, 2019-11-29:
I don't have any thermostat to test them. I can glue such sheet to v2 and see how it performs.

hartytp, 2019-11-29:
Let’s do that.

In practice I think we’ve done all we can to improve this design so I’m happy to finalise the pcb here

gkasprow, 2019-11-29:
fan supplies are routed far away from the ADC and they do not form ground loop.

hartytp, 2019-11-29:
@gkasprow i think this is all done so you can close unless you think I missed something

gkasprow, 2019-11-29:
I drafted new release 
https://github.com/sinara-hw/Thermostat/releases/tag/v2.0rc4

closed 2019-11-29 by gkasprow