sinara-hw/Thermostat #50: 5V on the 3V3 rail

hartytp, 2019-09-23:
@gkasprow out Thermostat card started playing up and the ADC suddenly became quite noisy. We noticed that the 3V3 LDO for the ADC was at 4V. It seems that the microprocessor digital outputs had hit 4.6V or so and were powering the LDO through the protection diodes.

After a bit more digging, it seems that the main 3V3 SMPS is now running at close to 5V.

I can't see any obvious way this could have happened. The main thing we've done to the board is to plug and unplug the PoE quite a few times during testing. I don't expect that to cause an issue, but could be related to https://github.com/sinara-hw/Stabilizer/issues/47 It's also conceivable that some clumsy probing at our end caused damage (although I do not believe that to be the case).

Okay if we send the board back to you for investigation?

gkasprow, 2019-09-23:
Yes please. Its better to figure it out before we go to new revision

Wyslano z Samsung w Plus
Pobierz aplikacje Outlook dla systemu Android<https://aka.ms/ghei36>

I can't see any obvious way this could have happened. The main thing we've done to the board is to plug and unplug the PoE quite a few times during testing. I don't expect that to cause an issue, but could be related to sinara-hw/Stabilizer#47<https://github.com/sinara-hw/Stabilizer/issues/47> It's also conceivable that some clumsy probing at our end caused damage (although I do not believe that to be the case).

gkasprow, 2019-09-25:
The DC/DC converter is shorted. The only explanation I see right now is transient on 5V rail which caused upper buck converter MOSFET damage.

hartytp, 2019-09-25:
Yes I wondered if that was the problem. Might also explain dead phy on stabilizer.

gkasprow, 2019-09-25:
Stabilizer generates 3V3 in a different way, using another type, single-stage converter which works up to 18V. On the Thermostat, the 3V3 rail is generated from 5V rail with converter that is able to survive max 6V.

hartytp, 2019-09-25:
Okay. Can you have a look at the 5V rail for transients when using PoE please?

hartytp, 2019-09-25:
I don’t have any more boards to test :(

hartytp, 2019-09-25:
Should we consider using the same smps as stabilizer next time to improve robustness?

gkasprow, 2019-09-25:
I was looking at it but did not detect any.
Is it possible that during plugging the PoE or 12V, somebody was touching the bottom side of PCB, under the DC Jack? There is a feedback divider so when sb toches it, may change division ratio and increase the 5V rail...
In case of the Stabilizer, it's much more difficult to access and touch the divider by chance

hartytp, 2019-09-27:
> Is it possible that during plugging the PoE or 12V, somebody was touching the bottom side of PCB, under the DC Jack? There is a feedback divider so when sb toches it, may change division ratio and increase the 5V rail...

I don't believe so, but it's always possible. Can we move these components to somewhere less accessible for the new revision?

Other than that, if you can't see any other issues with the design, let's close for now and reopen if this happens on another board.

Can you send us the repaired board back please so we can finish testing?

Thanks again for the work on this project @gkasprow !

gkasprow, 2019-10-26:
I moved the components to the top layer.

hartytp, 2019-10-26:
Okay. I have no other ideas for what could have caused this so let’s close for now

closed 2019-10-26 by hartytp