sinara-hw/Thermostat #51: PoE turns off and on

hartytp, 2019-09-25:
@gkasprow we saw this on one of our thermostat boards. I'll ship it to you for investigation.

hartytp, 2019-09-25:
It did stabilize and stay on eventually, but took quite a while for the switch to stop registering PoE faults

gkasprow, 2019-09-30:
in my lab it works fine
Here is the plot. Blue - 12V
green - PoE voltage before DC/DC
[image]

gkasprow, 2019-09-30:
The PoE voltage was measured with diff HV x100 probe

hartytp, 2019-09-30:
There is clearly something marginal somewhere so it's only particular combinations of boards and switches that fail to work.

Shall I send you the switch as well?

hartytp, 2019-09-30:
Actually, maybe first try it with the switch that @sbourdeauducq sent you? (Or see if you can reproduce this using the same stabilizer and swtich combination that he had?)

gkasprow, 2019-09-30:
I still didn't get the package from @sbourdeauducq

hartytp, 2019-09-30:
Okay, no problem. I'll send you the switch we used to reproduce this issue with. It's in use now, so will be a few days before I ship it.

gkasprow, 2019-09-30:
Anyway, I go to ESTEC for space & quantum conference, will be back on Friday.

hartytp, 2019-09-30:
Aah, ok. In that case, can you send me the Thermostat back and I'll do the tests.

Which tests would you like to see to debug this issue?

gkasprow, 2019-09-30:
I have with me only one with broken CPU and DC/DC which will pass to Technosytem for reparation tomorrow.

hartytp, 2019-09-30:
And broken ADC (the CPU applied 5V to the ADC via the digital outputs). So, most of the PCB is dead :(

I think you have two I sent you...one with the broken 3V3 SMPS (but working PoE) and one with a working SMPS but dodgy PoE.

gkasprow, 2019-09-30:
yes, but one is at the WUT and I'm leaving tomorrow.

hartytp, 2019-09-30:
Okay, well I need that one to test PoE as the other one doesn't display this issue. No worries if you can't ship it, we can wait until you get back.

hartytp, 2019-10-05:
Can you ship this back to me so I can debug?

gkasprow, 2019-10-05:
Yes, I will pass it to TS on Monday.

closed 2019-10-26 by gkasprow