sinara-hw/Thermostat #52: JTAG unreliable

hartytp, 2019-09-25:
On the board we shipped back to you (with the PoE issue) the JTAG is also unreliable and seems to die, requiring a power cycle, quickly. I believe that @astro and @sbourdeauducq also have this problem on their board.

Not sure if this is an issue related to the current microprocessor or if it's a general issue with the board.

I'm using an ST-Link adapter.

sbourdeauducq, 2019-09-25:
This could be an issue with the ST-Link adapter itself and not Thermostat. AFAIK programming became reliable after we started using a Olimex JTAG probe instead.

hartytp, 2019-09-25:
Okay, I'll try that (need to buy an adapter). However, I'm still surprised that this worked perfectly for the other board we had, but we have so many issues on this board. Same ST-Link adapter etc.

hartytp, 2019-09-25:
(unless it's something dumb like a driver issue/openocd version, since we're using a different computer for the new board)

sbourdeauducq, 2019-09-25:
Maybe because we kept it running for days?

sbourdeauducq, 2019-09-25:
And note that the ST-Link uses SWI and not JTAG.

gkasprow, 2019-10-26:
I think I cannot do much here. Let's see if the issue persists with the STM32 version of the Thermostat.

closed 2019-10-26 by gkasprow

hartytp, 2019-10-26:
@sbourdeauducq didn’t you find that a different programmer worked fine?

sbourdeauducq, 2019-10-26:
The olimex cable (JTAG and not SWI) works better.