sinara-hw/Thermostat #53: Review microprocessor connections

hartytp, 2019-10-26:
Can someone sign off on these before the next round (eg @jordens @astro @sbourdeauducq )

gkasprow, 2019-10-26:
the STM32CubeMX project is [here](https://github.com/sinara-hw/Thermostat/blob/master/STM32_F427_config.ioc)

gkasprow, 2019-10-26:
SPI assignments:
SPI2: ADC
SPI3: extension header
SPI4: DAC0
SPI5: DAC1

Timers
TIM1_CH1: PWM_maxIPos0
TIM1_CH2: PWM_maxIPos1
TIM1_CH3: PWM_maxINeg0
TIM1_CH4: PWM_maxINeg1
TIM3_CH1: PWM_maxV0
TIM3_CH2: PWM_maxV1
TIM4_CH1: ADC CLKIO
TIM4_CH1: ADC SYNC

ADC assignments:

ADC1_IN0: TEC Vref0
ADC1_IN3: TEC Vref1
ADC1_IN4: DAC_FB0
ADC1_IN5: DAC_FB1
ADC1_IN6: ITEC0
ADC1_IN8: ITEC1
ADC1_IN9: extension header.3
ADC1_IN10: P5V0 sense
ADC1_IN12: TEC_U_meas0
ADC1_IN13: TEC_U_meas1

UART1: header
I2C1: EEPROM

astro, 2019-11-11:
I cannot find `RMII_TXD0`. According to the [Datasheet](https://www.st.com/resource/en/datasheet/stm32f427zi.pdf), the only possible pin seems to be `PB12` which is used for ADC.

gkasprow, 2019-11-13:
well, the STM32CubeMx says that PG13 is also possible
[image]

astro, 2019-11-14:
Oh, I didn't recognize the second rectangle as the other half of the MCU. Continuing review...

astro, 2019-11-15:
Looks good to me.

hartytp, 2019-11-18:
@astro if you're happy with this, please close the issue. Thanks!

hartytp, 2019-11-18:
@gkasprow my understanding is that the microprocessor has 3 ADCs, with muxes to give 24 channels. We should make sure to connect Vref0 to the same ADC as DAC_FB0, same for channel 1, to maximise accuracy.

gkasprow, 2019-11-20:
Vref0 
[image]

DAC FB0
[image]

Vref1
[image]

DAC FB1
[image]

so we can assin all these inputs to ADC1 or ADC2.

closed 2019-11-20 by gkasprow