sinara-hw/Thermostat #54: v2.0 schematic review

hartytp, 2019-11-01:
@gkasprow is there any chance of getting a schematic rc release of Thermostat for Monday? I'd really like to finish my review.

hartytp, 2019-11-18:
@gkasprow okay, I believe all testing and reviews are now complete so we can finish the new hw rev.

What do you think the timeline is for getting these boards off to manufacture? Since the changes all seem to be quite small, hopefully we can do this quickly?

closed 2019-11-18 by hartytp

gkasprow, 2019-11-18:
Most of the issues are already implemented. Give me 2 days to implement them all.

hartytp, 2019-11-18:
Perfect! Thanks

hartytp, 2019-11-28:
ping @gkasprow can we get this sent off to manufacture on Monday?

hartytp, 2020-02-18:
@gkasprow I hear that the new boards are ready. Once you've smoke tested, can you send one of our board to M-Labs asap so they can finish the firmware? Thanks!