sinara-hw/Thermostat #55: Significant Coil Whine

pathfinder49, 2019-11-08:
Driving One TEC channel at maximum produces loud coil whine from the area of L2_1, L4_1, L2_2, L4_2. Might be worth checking everything is within spec.

Driving both TEC channels at maximum produces a louder whine followed by a shutoff after a few seconds.

This was observed when powering the board from the 12 V jack.

hartytp, 2019-11-08:
> Driving both TEC channels at maximum produces a louder whine followed by a shutoff after a few seconds.

I suspect the shut-off is a separate issue to the whine.

If the whine depends on the total power then it's possible it's the 5V SMPS rather than the TEC driver coils...

pathfinder49, 2019-11-08:
The inductors are also getting hot to the touch. L4_x seem to also be getting hotter than L2_x.

gkasprow, 2019-11-08:
you can observe the 5V rail to see if there are oscillations.
Shut off can be caused by thermal protection of the buck converter.

hartytp, 2019-11-09:
aside from the whine, the TEC driver inductors get scorchingly hot.

To do: review this part of the design and see if we want to make any changes. FWIW those inductors look pretty nice and seem to have an appropriate current rating so there may not be much we can do.

hartytp, 2019-11-18:
@gkasprow please can you review all coils/CMCs used in this design and check you're happy with them. 3A doesn't seem like such a high current to me, so I'm surprised the whine is this bad. Still, once Thermostat is in a box no one will notice, so if you're happy with the current parts feel free to close

gkasprow, 2019-11-21:
if it is audible, it means that there is some low-frequency component that is modulating the Peltier current.
Ceramic capacitors can also produce sound, especially ones with X7R  dielectric.
I'd measure the LF AC component of the TEC voltage to figure out the frequency range.

hartytp, 2019-11-21:
Okay, I'll try to find time to stick a scope on the TEC voltage tomorrow. @pathfinder49 is everything still set up for me to do this quickly?

pathfinder49, 2019-11-21:
I've disassembled things, but everything should still be in the same place.

hartytp, 2019-11-22:
@gkasprow we had another look at this issue. It appears that when the TEC driver gets close to saturating it starts cycle skipping. In our tests, one of the FETs was on almost constantly, and was only turning off briefly every ~400us. So, the switching frequency was much lower than designed.

Not sure if the low impedance of the inductors at this frequency was causing large current spikes to flow through the capacitors, or if the lower frequency just made the dynamics audible. Either way, turning the set current down a little so that the driver stopped cycle skipping the noise appeared to go away immediately.

As for which components were actually whining, it's hard to say. We tried the old "straw in the ear" technique to localize the whine, but it's hard to pin down precisely. I thought that the loudest whine was around the 5V SMPS. I think it was the ceramic capacitors at the SMPS output, but it could also have been the inductor inside the SMPS. Hard to say. There may also have been some capacitor/coil whine at the TEC driver output, but it's hard to say.

hartytp, 2019-11-22:
@gkasprow I don't think there is anything concerning here, so I'm closing.

If you're worried about the inductors/capacitors whining when the TEC driver gets close to saturation then please reopen.

closed 2019-11-22 by hartytp

jordens, 2019-11-22:
If it also does cycle skipping near the current zero crossing, that would be problematic as would make it electrically noisy for small thermal loads (i.e. laser diode off and near room temperature).

hartytp, 2019-11-22:
It doesn’t do this at 0 current. It behaves nicely as described in the data sheet (both FETs run at 50% duty cycle to give 0V differential across the load). This is purely an issue at saturation afaict

jordens, 2019-11-22:
I see. Nice.