sinara-hw/Thermostat #56: Change in ADC offset with TEC current

hartytp, 2019-11-09:
Data from here: https://github.com/sinara-hw/Thermostat/issues/46#issuecomment-551932456

[image]

@pathfinder49, do you have the raw data for that?

@pathfinder49 do you agree with the following statements based on the data you gathered?
1. Regardless of which TEC the current is applied to, the offset change is bigger for channel 1 than channel 2
2. (1) strongly suggests that the issue isn't related to the reference voltage or a supply rail, since that would presumably be common-mode to both channels
3.

hartytp, 2019-11-09:
(previous comment updated)...

hartytp, 2019-11-09:
Here is my theory: it's the ESD diodes we added very slightly rectifying the pickup. NB there isn't an EMI filter before the diodes, so any RF pickup reaches them unfiltered. 

[image]

@pathfinder49 can you quickly test this next week by trying something like:
- Add some large ceramic capacitors (100nF or even 1uF) here...

[image]

Or just remove D17 D18 D19 and D20

hartytp, 2019-11-09:
If that fixes the problem then the suggestions for the new revision are:
1. Be more careful with thermistor input trace routing (we now have 4 layers so can do this via internal screened layers)
2. Put an EMI filter before the diodes

pathfinder49, 2019-11-10:
> Data from here: [#46 (comment)](https://github.com/sinara-hw/Thermostat/issues/46#issuecomment-551932456)
> 
> [image]
> 
> @pathfinder49, do you have the raw data for that?
> 
> @pathfinder49 do you agree with the following statements based on the data you gathered?
> 
>     1. Regardless of which TEC the current is applied to, the offset change is bigger for channel 1 than channel 2
> 
>     2. (1) strongly suggests that the issue isn't related to the reference voltage or a supply rail, since that would presumably be common-mode to both channels
> 
>     3. For driving channel 0, the effect is larger when the TEC is positive than when it's negative.
> 
>     4. One interpretation of (1) and (3) would be that the effect is larger when the current flows through L2, and largest when it flows through L2_1. This could well be an indication that pick-up on the ADC lines (which run along the bottom-edge of the topside of the PCB) is the issue (L2_1 is right next to those lines).
> 
>     5. There is also a slow baseline drift in the measurement which is presumably just the board heating up. We will measure the temp co, but I'm not too surprised to see sub-mk changes in the measured reading when the board starts to get that hot. I suspect that even some mild forced air cooling would make this a non-issue...
> 
>     6. The change in ADC offset with TEC current is not linear.
> 
> 
> [image]

I'd mostly agree... 

1. ch1 has a bigger response than ch0 to TEC0. However the TEC1 response is similar.
6. I will plot this tomorrow. However some quick calculations indicate the response is linear.

pathfinder49, 2019-11-10:
> Here is my theory: it's the ESD diodes we added very slightly rectifying the pickup. NB there isn't an EMI filter before the diodes, so any RF pickup reaches them unfiltered.
>
> [image]
>
> @pathfinder49 [...]

I'll have a go at that.

pathfinder49, 2019-11-11:
The ADC offsets appear consistent with a response proportional to TEC current. However, the gradient appears to be different for different sign currents and for different TEC/ADC channels,
[image]

pathfinder49, 2019-11-11:
On a second look, the TEC1 response is not linear. (editing plot above to include starting position)

hartytp, 2019-11-11:
Ok. Well, either way, let's figure out the cause.

I see two possibilities: ground loops or diode rectification.

We can quickly check rectification by adding capacitors on the diode inputs (see above). 

To check for ground loops, we can rotate the input resistors to ground by 90deg (or lift them from the pads) and add a wire to the ADC ref- pin. That way we have a Kelvin connection so aren't sensitive to ground currents.

I bet one of those two will kill this issue...

closed 2019-11-18 by hartytp