sinara-hw/Thermostat #58: kelvin-connect the TEC divider to ground near the reference

hartytp, 2019-11-11:
to minimize sensitivity to ground loops...

hartytp, 2019-11-12:
Let's also use star-routing on the Vref+ TEC lines to avoid cross-talk between the two TEC lines. Basically, have stars on both TEC+ and TEC- that go to the ADC REF+ and REF- lines.

gkasprow, 2019-11-20:
do you mean this ground?
[image]

gkasprow, 2019-11-20:
TEC+ and TEC- are routed to the diff amplifiers. And the divider ground should indeed be connected as close to CPU as possible.

gkasprow, 2019-11-21:
OK, so I implemented star connection for both ref and divider GND
[image]
[image]

hartytp, 2019-11-21:
@gkasprow sorry! I meant that we should kelvin-connect the thermistor v_ref+ and v_ref- to near the ADC. Not the TEC driver.

hartytp, 2019-11-21:
i.e. what I want to do here is (a) fix the ground loop that caused the ADC offset to vary with the TEC current (b) minimize the coupling between the two ADC channels.

gkasprow, 2019-11-21:
In such case, one has to make sure that return current does not pass via sensitive analog circuits.
It should be enough to add a plane cutout that will make the current loop much smaller.

[image]

I also added Kelvin connections for thermistors. I used net ties to make sure the vref- won't be shorted to GND by coincidence.

gkasprow, 2019-11-21:
Vref+
[image]
Vref-
[image]

closed 2019-11-21 by gkasprow