sinara-hw/Thermostat #59: thermal management

hartytp, 2019-11-12:
Ultimately there isn't a huge amount we can do to stop the board heating up when run without forced air cooling at max TEC current. I suspect that the long-term plan will involve documentation (if you want good stability at high power then used an enclosure with good ventilation) and software fixed (e.g. maybe power limiting dependent on the enclosure).

Some things we can consider:
- add heat sinks. On components that get hot? or just under the ADC?
- add an I2C thermometer (I know there is one in the ADC, but using that costs us loop BW)
- add a fan control header hooked up to the microprocessor so one can run a 12V/5V/whatever fan

pathfinder49, 2019-11-12:
I've Stressed the TEC by drawing 3A on both channels and was not able to repoduce the immediate shutdown. However componets reached 130°C over 2 minutes at which point I shut down the board.

I will post images once I can get them off the camera.

hartytp, 2019-11-12:
Ouch! Bearing in mind that you're looking at case temperature, not die temperature, you must have been close to putting some components into thermal shutdown!

jordens, 2019-11-12:
See #38 for the analysis and several other recommendations for improved thermal management.

jordens, 2019-11-13:
Is it primarily the inductors getting hot or the fets in the switchers?

pathfinder49, 2019-11-13:
Front of the PCB
[image]

Back of the PCB
[image]

pathfinder49, 2019-11-13:
Here some images with PoE. These are after some time driving 2 A on both TEC channels.

Overview (front)
[image]
Overview (back
[image]
)

hartytp, 2019-11-13:
> Is it primarily the inductors getting hot or the fets in the switchers?

Which switcher are you asking about? Poe? 5V? or TEC driver?

hartytp, 2019-11-13:
Summarising a bit:
- with the board lying flat on a desk (no airflow on the bottom side, which has the TEC driver)
- The TEC drivers got the hottest. I saw 120C at one stage with them running at ~2.5A each
- Close second was the 5V SMPS and the TEC inductors, which all topped 110C. NB the inductor in the [5VSMPS](http://ww1.microchip.com/downloads/en/DeviceDoc/20005571A.pdf) is internal. 
- With PoE, the PoE module got hot, but not quite as hot as the main board.

@pathfinder49 please can you post photos we took of the PoE module?

jordens, 2019-11-13:
The three. They should be about equal in loss. And from the looks of it it's the FETs. That is nice for conduction cooling/heat sinks.

hartytp, 2019-11-13:
The FETs, as in the ones inside the TEC driver chip itself?

jordens, 2019-11-13:
Yes.

hartytp, 2019-11-13:
Anyway, as you say, the heat is distributed somewhat evenly around the board. NB the ADC was getting to 60C, so one can expect poor stability when running at higher powers without pretty good active cooling.

In the next revision, with more layers, we'll get better heat spreading. But, I still think that a heat sink would be a good idea.

jordens, 2019-11-13:
Is this with 500 kHz or with 1 MHz switching frequency by the way? This should make a bit of a difference.
Conduction cooling would directly use the enclosure as the (much bigger) heatsink.
It looks like we are not far away from getting respectably high sustainable TEC currents out of this. That's really nice.

hartytp, 2019-11-13:
> Is this with 500 kHz or with 1 MHz switching frequency by the way? This should make a bit of a difference.

Good question, I'll have to check and get back to you.

> Conduction cooling would directly use the enclosure as the (much bigger) heatsink.

How would we get conduction to the enclosure?

hartytp, 2019-11-13:
> Is this with 500 kHz or with 1 MHz switching frequency by the way? This should make a bit of a difference.

From a quick skim over the firmware, I do not believe this pin is currently actively driven. Assuming the microprocessor output is high-Z. Looking over the [datasheet](https://datasheets.maximintegrated.com/en/ds/MAX1968-MAX1969.pdf) I don't see any mention of a pull-up/pull-down on that pin. So, it's anyone's guess what we used.

It's a shame we didn't spot that before testing, but we're running out of time for testing now so I propose we leave it there. I don't think there are any decisions we will make conditioned on the outcome, so makes sense to finish testing on the new hw rev.

pathfinder49, 2019-11-13:
> From a quick skim over the firmware, I do not believe this pin is currently actively driven. Assuming the microprocessor output is high-Z. Looking over the [...]

I previously saw unconfigured pins as low on thermostat.

jordens, 2019-11-13:
Conductive cooling builds low thermal resistance paths from the sources through copper planes to the edges of the board. There you expose the planes and make large area thermo-mechanical contact to the enclosure/structure with screws directly or with wedges. This is how you do thermal management in military or satellites where you can't rely on air.
For us this just means making the thermal path to the edges fat and then ensuring that the enclosure can make good thermal contact to the board edges. Alternatively make direct thermal contact to places of the board close to the heat sources.

gkasprow, 2019-11-13:
In our case, the simplest thermal interface would be a piece of a thick copper-polymer sheet  (gap filled pad) with thin AL foil on top to enable easy sliding into the enclosure. It is elastic and non-conducting but contains even 80% of copper. One can glue it to the PCB bottom and maybe to the top of inductors. The ICs have much smaller Rth to the PCB than to the case, so the gap filler should be much larger, maybe the size of the board.

One can buy them in [Farnell](https://pl.farnell.com/w/c/zarzadzanie-chlodzeniem-i-temperatura/materialy-przenoszace-cieplo/materialy-termoprzewodzace?przewodnosc-termiczna=3w-m-k|3.5w-m-k|4w-m-k|5w-m-k|6w-m-k&st=gap%20filler)

gkasprow, 2019-11-13:
The LM75 thermometer is already installed on the PCB. I placed it close to CPU but can move it between TEC drivers and 5V SMPS. It will disable the main supply when 80 deg is exceeded.

gkasprow, 2019-11-13:
@jordens inductors are placed just above the ICS which dissipate head to the PCB and heat the inductors. So they are thermally linked together.

jordens, 2020-10-29:
The poe module has some recommendations for heat management. Especially useful heat sinks and thermal pads at the end of their app note: https://silvertel.com/images/appsnotes/ANX-POE-Thermal-Considerations.pdf

gkasprow, 2020-10-30:
I managed to complete the setup. 
[image]

gkasprow, 2020-10-30:
After a  few minutes of running with 3V at both outputs the circuit enters thermal protection mode

gkasprow, 2020-10-31:
The resutls are [here](https://github.com/sinara-hw/Thermostat/issues/73#issuecomment-719834216)

closed 2020-11-02 by hartytp