sinara-hw/Thermostat #61: ref 2V5

hartytp, 2019-11-18:
This currently looks to be driven by both the 24-bit ADC and the MCP1525T.

AFAICT it would be better to power the microprocessor ref from a 3V3 reference anyway: currently we have separate dividers for the DAC output going to the TEC driver/microprocessor. That will limit the accuracy of the TEC driver calibration. If we used a 3V3 microprocessor reference then we could scrap the divider for the microprocessor... 

[image]

[image]

hartytp, 2019-11-18:
Changing the microprocessor reference will also require changing the resistors used for the TEC current measurement. 
[image]

gkasprow, 2019-11-20:
The assembly variant makes the MCP1525 DNP so the CPU ADC is driven from the 24bit ADC.
We don't have any 3V3 reference. I would have to replace the MCP1525 with a 3V reference. I'm not sure if we gain a lot here. We will add yet another reference voltage that is not related to ADC reference nor DAC reference.

hartytp, 2019-11-21:
I'd like to be able to have the same divider on the DAC outputs going to the microprocessor and TEC driver so we don't need to think about tolerances on the dividers.

I'd be happy with:
- generate a 3V0 (or 3V3) microprocessor reference by dividing the 5V0A rail with a resistive divider (IIRC the microprocessor has a ref buffer so this can be relatively high-Z)
- use a filtered version of the 3V3 SMPS rail for the microprocessor ADC reference.

hartytp, 2019-11-22:
@gkasprow do you think we can get v2.0 ready to ship on Monday?

hartytp, 2019-11-25:
ping @gkasprow when can we get this sent off?

gkasprow, 2019-11-29:
I didn't look here for a while. Next time just write an email. All github issues mails land in dedicated folder. Yes, we can produce it on Monday.

hartytp, 2019-11-29:
Great. Can you produce an RC before that so I can sign off on the final changes?

gkasprow, 2019-11-29:
the reference input in STM32F4 is not buffered
[image]

the easiest is to add opamp buffer (LMV321ILT) that we already use and remove MCP1525 to make space

gkasprow, 2019-11-29:
[image]

gkasprow, 2019-11-29:
the DAC output would look like this.
[image]

The 1k6 at the output isolate DAC from ADC current pulses caused by ADC sampling circuit

hartytp, 2019-11-29:
Lgtm

closed 2019-11-29 by hartytp