sinara-hw/Thermostat #66: Thermal sheets

hartytp, 2019-11-29:
Try them with v 2.1 to see if we can conduct heat from the tec drivers to the enclosure

gkasprow, 2019-11-30:
@hartytp how many v2 boards shall we produce?

hartytp, 2020-02-28:
@gkasprow Thermostat v2.0 has arrived from TechnoSystem. The enclosures look great! Can't wait to try it.

Currently it doesn't have the thermal sheeting we previously discussed (to improve the heat conduction from the PCB to the case).

Can you recommend a PN for the thermal sheeting we should use and give some advice on how to install it/what we're trying to achieve (i.e. is the idea to just uniformly coat the PCB and also get a thin layer that will squish in the slot where the PCB mounts to the case to improve conductivity?).

Once the new FW is ready, we'll test the thermal performance both with and without the sheeting and see if it's worth it. If it is, the we should add it to the BOM.

gkasprow, 2020-02-28:
I didn't see them. I'll ask TS for additional unit I could test. I have such sheets in my lab.
I'd try to use one of[ these](https://pl.farnell.com/w/c/zarzadzanie-chlodzeniem-i-temperatura/materialy-przenoszace-cieplo/materialy-termoprzewodzace?grubosc=1.524mm|2mm|2.032mm|2.54mm|3mm|3.175mm|3.18mm|3.2mm|4mm|4.1mm|5mm|5.08mm&range=inc-in-stock), just choose one with right thickness. Glue Alu foil on top to make the PCB  insertion feasible.

hartytp, 2020-02-28:
Aah cool! Ok you have a play with it and tell us what you think we should do.

hartytp, 2020-02-28:
Let me know if you need me to send you a Thermostat

gkasprow, 2020-02-28:
I have one thermostat, just need an enclosure with panels. I will get them from TS on Monday. From you, I need a firmware that is able to provide the conditions I need. I can also attach tiny PT1000 to measure temperatures.

hartytp, 2020-02-28:
Right now there isn’t any firmware for the v2 hardware (STM micro). M-Labs are planning to port their firmware. We can put this testing off until the new firmware is ready if it’s easier, there isn’t a rush on this...

gkasprow, 2020-02-28:
OK, I can generate some simple code with CubeMX and make PWMs running to dissipate some power.
However if Mlabs firmware is coming soon, it's better to wait.

sbourdeauducq, 2020-03-24:
The board gets surprisingly hot, with only 1 TEC connected.

hartytp, 2020-03-24:
I assume it's comparable to the v1.0 hardware that we characterized before (several sets of ~85% efficient SMPSs add up to quite a bit of power dissipation)? See the thermal images we posted.

Once the firmware works well enough it would be great for @gkasprow to have a play with thermal conductive sheets as discussed above.

gkasprow, 2020-03-24:
In a few days I will receive the multi-point thermometer so will do detailed analysis with thermal sheets and without them.

hartytp, 2020-03-24:
Great!

gkasprow, 2020-10-11:
I wrote a half a year ago about the multi-point thermometer. I received two such boxes a few days ago ...
[image]

gkasprow, 2020-10-31:
@hartytp look [here ](https://github.com/sinara-hw/Thermostat/issues/73#issuecomment-719834216)

closed 2020-11-02 by hartytp