sinara-hw/Thermostat #68: star ground

hartytp, 2019-12-01:
@gkasprow I don't have altium, so it's hard to follow the layout. But, just to check: the thermistor ref gnd should be tied to ADC pin 2 (ref -) but on the schematic it looks like it's tied to C54. Why is this?

hartytp, 2019-12-01:
@gkasprow can you sketch out the routing of the ref and ref GND to the thermistors for me please?

hartytp, 2019-12-01:
FWIW I don't understand why IC14 uses the ADC 2V5 reference and REF GND. Why not just use the TEC ref output? Would simplify the routing and ensure that we don't accidentally couple noise form the TEC to the ADC.

Not a big deal and probably not worth changing now, I just don't understand why it was done -- although, I suspect it was a mis-communication between us.

hartytp, 2019-12-01:
@gkasprow thanks again for all the work on this.

It's hard to follow the layout from the PDFs, but the bits I can follow look really nice. Good work. Will be very curious to see what happens when we stabilise one of our cavities. <100uK RMS?

gkasprow, 2019-12-01:
[image]

gkasprow, 2019-12-01:
I used net ties to make sure the GND is not shorted to GND. Net tie creates a different net names while providing physical connection. ADC ref- goes to GND so it is the same net. I can change on the schematic but it does not change anything on PCB.

hartytp, 2019-12-01:
Okay that’s pretty good

I would prefer if possible to have the star ground for the ref - right at the ADC ref - pin rather than at the ep. But that’s close

gkasprow, 2019-12-01:
OK, no problem. It that make you sleep better... :)

hartytp, 2019-12-01:
Your call. Either route as a proper star ground right to the pin or close now if you think it’s not worth it (in any case the new layout with gp cuts and more layers should make these ground issues less of a provlem)

hartytp, 2019-12-01:
Thanks! The thing killing my sleep right now is booster

gkasprow, 2019-12-01:
[image]

hartytp, 2019-12-01:
Perfect

closed 2019-12-01 by hartytp

gkasprow, 2019-12-01:
True, IC should use ADC reference. 2V5 does not much sense here. It is residue from previous revision.

gkasprow, 2019-12-01:
I mean IC14 and CPU ADC

hartytp, 2019-12-01:
Indeed. I think that right now the only thing that uses the ADC ref is that opamp and that should probably use the tec ref for convenient (the tec voltage measurement doesn’t need to be particularly high accuracy).

hartytp, 2019-12-01:
If you can be bothered then fix it as it makes the design neater, but it’s not crutical

gkasprow, 2019-12-01:
fixed