sinara-hw/Thermostat #70: PoE/12V switch

hartytp, 2019-12-01:
One thing I forgot to ask before: in Stabilizer you have a switch to protect against power being supplied by PoE and the barrel connector at the same time. Why isn't that needed here?

hartytp, 2019-12-01:
Also, with 22uF the time constant is quite high. Does it need to be that high with the optocoupler arrangement? We just need to be careful not to let the firmware turn the TEC power to max too quickly before the resistor is shorted out.

Might be best to just DNP C136

gkasprow, 2019-12-01:
True, we probably don't need C136 here.
I'm not 100% if we need the power switch. Its purpose was to protect the PoE module optocoupler. I don't know if the PoE module regulation loop won't try to dump too high current to the feedback optocoupler causing it to fail after some time. It's pure speculation.

hartytp, 2019-12-01:
I don’t feel too strongly but it feels odd to have it in one design but not the other

jordens, 2019-12-01:
It would be very useful to give the cou knowledge about whether it's powered by poe or 12 v. See the thermal management analysis. Do we have that on both stabilizer and Thermostat?

gkasprow, 2019-12-01:
I've just checked on the module that there is 1k2 0603 resistor between 12V and TL431.
This means that with the external voltage applied that exceeds 12V, the optocoupler LED will get max 8mA. So we don't need this switch here as well as in the Stabilizer. The resistor will dissipate 64mW.

gkasprow, 2019-12-01:
It's easy on Stabilizer - need just one resistor to implement it. On Thermostat we would need to add such switch.

gkasprow, 2019-12-01:
OK, I added the ideal diode switch and 2 resistors. The CPU will know what the source is.

closed 2019-12-01 by gkasprow

hartytp, 2019-12-01:
Awesome, thanks @gkasprow !

hartytp, 2019-12-01:
BTW I think the schematic is now out of date. When you create the production files, can you load a final release?

gkasprow, 2019-12-01:
done
https://github.com/sinara-hw/Thermostat/releases/tag/v2.0rc5