sinara-hw/Thermostat #73: V2.0 testing

hartytp, 2020-02-28:
On top of the tests we performed on v1.0, I suggest the following tests for v2.0:
- [ ] Glue a couple of thermistors on the places we expect to be hottest (e.g. TEC driver). Run the unit at full power inside the case and look at the change in temperature. Do this with/without the thermal sheeting and using both PoE and a barrel connector.
- [ ] While taking the above measurement, monitor the ADC reading with a precision 10k resistor in place of a thermistor. Look for both step-changes in the ADC reading when the TEC is turned on and slower (thermal) changes as the unit heats up
- [ ]  Temperature stabilize a commercial optical cavity. Monitor the temperature using a witness thermistor and Keysight 7.5 digit DMM.

gkasprow, 2020-03-01:
Measurement of temperatures inside of the enclosure is simple from one side and tricky from the other side. Usually, I use a set of tiny PT1000, thin wires, DMM and rotary switch. But this is problematic and takes a lot of time. To make my life easier, I designed a [simple box](https://github.com/gkasprow/temperature-logger/wiki) with various configurations of [sensors](https://github.com/gkasprow/temperature-logger/issues/1) that do exactly what we need. It took me 2 days including PCB, enclosure, 3D printing, and basic firmware. But this will pay off. 
The box will dump measurements to the SD card, serial port or via the MQTT.

gkasprow, 2020-03-01:
I plan to produce 5 units for me ASAP. If sb wants, I can make a few more.

gkasprow, 2020-05-13:
V2 photos 
[image]
[image]

astro, 2020-09-13:
I'm using the green output connectors which are usable through the enclosure. The labelings for TEC1/TEC2 on the board, and 0/1 on the enclosure should be swapped.

gkasprow, 2020-09-13:
@astro open dedicated issue, please.

gkasprow, 2020-10-11:
I finally got the multi-channel PT1000 measuring box, TECs, sensors, thermal sheets. I'm playing with parameters and Python scripts right now.

gkasprow, 2020-10-30:
after playing with PoE switch the Thermostat stopped responding.. The PHY LEDs are on continuously, I see a startup message on the USB port. After some time the PHY LEDs go off.

gkasprow, 2020-10-30:
I replaced the PHY and Ethernet works again.

gkasprow, 2020-10-30:
I was using a 15W PoE switch before and nothing happened. Today I switched to 30W one and after a few plug-unplug cycles the PHY stopped to work

jordens, 2020-10-30:
This looks so much like the issue on Stabilizer that it makes a PCB manufacturing error a bit unlikely (those are two different boards), right?

gkasprow, 2020-10-30:
Exactly. The PoE and PHY circuit is identical in both cases.

gkasprow, 2020-10-30:
the PHY Ethernet signals are protected using TVS diodes. Anyway, I will probe them with the scope during plug/unplug action.

gkasprow, 2020-10-30:
the thermal test parameters:
tec = Client() #(host="localhost", port=6667)
tec.set_param("s-h", 0, "t0", 32651)
tec.set_param("s-h", 1, "t0", 32651)
tec.set_param("s-h", 0, "b", 3977)
tec.set_param("s-h", 1, "b", 3977)
tec.set_param("pwm", 0, "max_v", 5)
tec.set_param("pwm", 1, "max_v", 5)
tec.set_param("pwm", 0, "max_i_pos", 4)
tec.set_param("pwm", 1, "max_i_pos", 4)
tec.set_param("pwm", 0, "max_i_neg", 4)
tec.set_param("pwm", 1, "max_i_neg", 4)
tec.set_param_short("pwm", 0, -3)
tec.set_param_short("pwm", 1, -3)
tec.set_param("report mode", "off")

gkasprow, 2020-10-30:
[image]

10h50 - power on, network connection. The board was warm due to previous experiments
10h59 - execution of the script listed above.
11h05 - board enters thermal protection mode. The LM75 has some hysteresis which causes oscillations and periodic disabling of the power stage. 
11h15 - a small fan (40x40x10, 12V, 40mA) was placed partially covering one endcap. Some fraction of the flow passes via the enclosure cooling the PCB.
11h24 - the fan voltage was set to 6V, 23mA.
11h37 - the fan voltage was set to 4V, 15mA
11h43 - the fan voltage was set to 3V, 12mA; the fan placement changed slightly
11h45 - the fan voltage was set to 6V, 23mA

The board consumes 27.5... 29W of power, depending on the temperature inside.

gkasprow, 2020-10-30:
the conclusion is:
- even very small airflow drastically improves working conditions of the board.
- we need to replace the inductors with ones with smaller losses
- let's try with thermal sheets, but I do not expect drastic improvement
- we can fit small 25x25x8 or even x10 fan to esisting design. Fan control circuit is already there.

gkasprow, 2020-10-30:
here are the raw data
[Uploading full power passive and fan.zip…]()

gkasprow, 2020-10-30:
I installed a sandwich of copper-filled thermal sheet, alu block, and again the sheet between inductors and the enclosure.
Another sheet was placed between the driver ICs and enclosure. Assembly of it is very problematic but let's see what we get

gkasprow, 2020-10-31:
It seems the thermal pads helped a lot. One of the enclosure sensors died during measurements, but we have two of them. It was misbehaving before.
[image]

gkasprow, 2020-10-31:
raw data are here
[TH pads full power passive and fan.zip](https://github.com/sinara-hw/Thermostat/files/5468293/TH.pads.full.power.passive.and.fan.zip)

gkasprow, 2020-10-31:
The PoE has its own thermal shutdown that activates after exceeding 95 degrees. 
[image]

gkasprow, 2020-10-31:
The AG5300 [datasheet](http://silvertel.com/images/datasheets/Ag5300-datasheet-smallest-30W-Power-Over-Ethernet-Plus-Module-PoEplusPD.pdf) says about 85degrees, I'm measuring the temperature on the inductor, not the IC, so that makes sense.

hartytp, 2020-10-31:
@gkasprow okay, so thermal sheets and a bit of forced air let the unit run at max power dissipation while remaining at pretty low temperatures. As you note, we may also be able to make some (small?) improvements by looking for lower loss inductors.

But, how practical are these changes? Do we think they will work in practice/production?

I wonder if the most practical solution here would be to say that the compact version is intended for lower powers (e.g. using some software limiting); higher powers should use the EEM version with forced air cooling. Obviously if we can find a nice solution for making this version work at max power that would be great, but we'd need to avoid it without expensive machining or fragile assemblies.

FWIW the use cases I have in mind for this are mainly quite low powers...

gkasprow, 2020-10-31:
Thermal sheets do not help that much in case of forced-air cooling. They lower the inductor temperature mostly. They are very problematic in assembly. So I recommend adding a small fan only. I ordered a few of them. It will fit easily.
[image]
Even with reduced airflow, where the fan noise is minimal, the temperatures are acceptable
[image]
At 2h25 I reduced the fan voltage. I will add PWM switch so the CPU will be able to regulate the fan speed.
I'm running it with 29W consumed from 12V rail.

gkasprow, 2020-10-31:
Machining of single side of the enclosure is cheap. 0.5mm precision is enough and the most expensive part is the precise fixing of the extruded part. Since this is not the case, the current impact would be neglidgible. Fan would be fixed by 2 self-tapered screws.

gkasprow, 2020-10-31:
we can also glue small heatsink on top of the inductors.

gkasprow, 2020-10-31:
I just removed the thermal sheets, and with reduced air flow the temperatures are:
[image]

gkasprow, 2020-10-31:
I'm curious why the output inductors heat that much. They have 54mOhm resistance. The CMC chokes at the output have 20mOhm and they are not heating that much, They are dissipating nerarly the same amount of heat because there are two coils.
I will add more channels to study this issue.

gkasprow, 2020-10-31:
They are thermally connected to the driver ICs, so it's quite possible that the inductors are simply acting as the ICs radiators :)

gkasprow, 2020-10-31:
It's easy to check if switching losses or conduction losses dominate in the indctors. If we lower the current to a very low value, the inductor should be cold.

gkasprow, 2020-10-31:
In steady-state we have 
[image]
which means the inductors are much hotter than the ICs. On the other side, The probe is glued to the top of the IC and to the inductor. And the RTh of inductor is much lower than plasitc package if the IC. We will need a tiy heatsink on top of the inductor anyway.

gkasprow, 2020-10-31:
reduction of output voltages from 3V to 1V drastically changes temperature distribution in the circuit. It looks like inductors do not heat due to switching losses.
[image]

gkasprow, 2020-10-31:
I attached a small heatsink to the inductors. It reduced the temperatures of:
- inductors by 15 degrees 
- driver ICs by 5 degrees
- PCB close to the drivers by 6 degrees.
- PoE by 7 degrees
- DC/DC by 4 degrees
It's a simple and low-cost solution so we should apply it. It obstructed the airflow so more air is passing closer to the PoE module and DC/DC converter, that's why their temperature also dropped.
This is steady-state
[image]

hartytp, 2020-11-02:
@gkasprow how do you plan to make the fan work with the small enclosure? Will you machine an opening in the enclosure?

gkasprow, 2020-11-09:
@hartytp I received the 35x35x10 fan. It fits nicely and leaves enough clearance for the air. I will machine the openings like in the drawing
[image]

I will do it on my CNC machine, install the fan and repeat all measurements.

hartytp, 2020-11-09:
Nice. What's the additional cost of machining the enclosure? Where is the air intake? Grill on one of the panels? Are you also adding heat sinks to lower the thermal impedance?

gkasprow, 2020-11-11:
The intake is above the fan. The outlet is on the side where TEC connectors are.
The cost of machining would be 2..3EUR @ 20pcs

hartytp, 2020-11-11:
:) that's awesome.

Did you get a chance to measure the temperatures?

dtcallcock, 2020-11-11:
Will you also test Thermostat in the Eurocard rack configuration (with and without forced air)?  Or is it just a safe assumption that if this config works, so will that?

hartytp, 2020-11-11:
yeah, my expectation is that in a rack with some decent forced air cooling this will be absolutely fine

gkasprow, 2021-02-03:
I will do tests. I'm still waiting for the machined enclosures. Once they arrive I will repeat tests in both configirations

gkasprow, 2021-03-18:
I finally received the enclosures. All fit perfectly. It's time to finish tests and make the release.
[image]
[image]

hartytp, 2021-03-18:
That looks nice! Will be interested to see how hot things end up, but it definitely looks the part.

Shame we loose the silk for the connectors, but I guess that's unavoidable.

gkasprow, 2021-03-18:
the fan is loud at max RPM, the airflow is much higher than I used during thermal tests. I already added PWM control circuit.
These new endcaps were 3d-printed but will be laser-cut and attached using screws.
The silk will fit under connectors
[image]

gkasprow, 2021-05-02:
Today I found some time and assembled the test suite. I connected the fan (12V version) to the 5V rail because there is no PWM circuit in v2.2. The fan spins slowly, it's less audible than my laptop in idle mode. However, it seems it is sufficient to cool down the electronics. Both outputs are running at 3A, one is 2.7V, another is 3V. The thermostat is supplied from the PoE switch. After 13mins it achieved the steady state.
[image]

gkasprow, 2021-05-02:
Moreover, the airflow inside is obstructed by 8 thin measurement cables and sensors.

gkasprow, 2021-05-02:
An observation - without the USB cable attached, the LAN does not start and the upper red LED is lit.

gkasprow, 2021-05-02:
Let's try it with a 12V fan supply. The noise dominates all the noise sources in my lab. The steady-state is achieved after 6 mins.
[image]

All temperatures dropped significantly. Only the TEC driver is 3 degrees cooler because it is on the bottom side of the PCB where there is no airflow.

gkasprow, 2021-05-02:
I think the results are fully satisfying and we can make a new v2.2 release. The fan and mounting screws are added to the BOM.

gkasprow, 2021-05-02:
The internal temperature sensor is placed just about the TEC drivers so we can use it to control the fan RPM and make sure we never exceed let's say 65 degrees. In most conditions that will result in low audible noise operation.

closed 2024-06-17 by maciejprzybysz