sinara-hw/Thermostat #78: Pull ups/pull downs

hartytp, 2020-05-26:
Let's do a review of the pull ups/pull downs before the next release.

For example: the DAC part number in the BOM does not specify the full DAC part, in particular whether it should power on to mid-scale/zero. If that's a deliberate choice to avoid availability issues then it's fine, but we should assume a random mixture of both parts. That means that if the shutdown is not pulled low at boot, we could (I think, but need to double check what the I/V lims do in the current design) end up driving max tec current until the DAC is first programmed. Might be worth adding a pull-down on the shutdown pin to ensure the TEC driver doesn't drive any current until everything has been configured.

gkasprow, 2020-10-11:
right, the full DAC PN is AD5680BRJZ-1500RL7 which has power on output state: 0
other versions of this DAC are not avaialble.
pull-down on the SHDN pin is a must.

gkasprow, 2021-03-15:
fixed

closed 2021-03-15 by gkasprow