sinara-hw/Thermostat #80: testing - connection issue

gkasprow, 2020-09-19:
I have an issue with communicating with Thermostat over Ethernet
I used a recent elf file which converted to dfu and flashed the chip.
It pings it at 192.168.1.2

jordens, 2020-09-20:
I haven't done much with thermostat yet. @astro or @hartytp or @pathfinder49 ?

hartytp, 2020-09-20:
We haven't touched v2.0 yet as we're waiting for a firmware rc.

gkasprow, 2020-09-20:
I'm referring to the connection instruction on the M-labs wiki. So probably it's a question to @sbourdeauducq

astro, 2020-09-20:
Known issue: https://git.m-labs.hk/M-Labs/thermostat/issues/2

I'm still researching its cause. Let me add the workaround to our Nix/Hydra build in the meanwhile...

gkasprow, 2020-09-20:
Let me know when you generate binaries that work with TCP

astro, 2020-09-23:
Seems to be fixed now.

gkasprow, 2020-09-23:
@astro can you generate binaries for me, please? preferably dfu file.

sbourdeauducq, 2020-09-23:
Binaries are automatically generated on our server whenever input repositories are modified. The latest ones would be https://nixbld.m-labs.hk/build/91375

astro, 2020-09-23:
A link to the always latest build is atop of the firmware's README.

astro, 2020-10-07:
(Resolved but I lack permissions to close this issue.)

closed 2020-10-07 by sbourdeauducq