sinara-hw/Thermostat #83: PoE kills PHY

gkasprow, 2020-10-31:
It was already discovered in Stabilizer and confirmed in THermostat that some PoE switches kill the PHY.
This is a known problem:
https://e2e.ti.com/support/power-management/f/196/t/861446
They solved it by adding TVS, but we already have them.

gkasprow, 2020-10-31:
Interesting APP note
http://ww1.microchip.com/downloads/en/AppNotes/00002157B.pdf

gkasprow, 2020-10-31:
It looks like we need to add four limiting resistors

gkasprow, 2020-10-31:
Anyway, this time the Thermostat survived plugging the RJ45. It may get damage during unplug due to transients. During life unplugging, a few hundreds of mA suddenly stops flowing via the isolation transformer. It's a lot of energy. Enough to damage the PHY

jordens, 2020-10-31:
A couple other things I found:

"Note: “Bob-Smith termination does not apply for Power Over Ethernet (PoE) applications."

[image]

From [TI](https://www.ti.com/lit/an/snla079d/snla079d.pdf).

And there seem to be special magnetics for PoE. I would have thought that our RJ45 is well equipped to handle PoE. But maybe worth checking.

gkasprow, 2020-10-31:
we are already using dedicated PoE transformer with passives.
[image]

gkasprow, 2020-10-31:
Yesterday I didn't switch off the PoE switch but was plugging and unplugging the RJ45 cable.
During plugging, there is no voltage on the contacts.
But during unplugging, one of the contacts always disconnects first causing an imbalance in the transformer.
So I suspect that the PHY gets damaged during the disconnect of RJ45.

gkasprow, 2020-10-31:
I connected the decent scope to Rx, Tx, AVDDPHY and tried hard plugging and unplugging the RJ45 and didn't manage to catch any transient that is higher than standard link FLP/NLP pulses.

gkasprow, 2020-10-31:
the PoE, 5V buck and 3.3V buck converters are stable. I did a step load test and the overshoot is lower than 100..200mV

gkasprow, 2020-10-31:
I mounted the broken PHY, but it does not have any signs of life. It does not emit even FLP/NLP pulses. I mounted back the working chip and it works again.

hartytp, 2020-11-02:
@gkasprow are you happy that with the different TVS and resistors you are no longer able to damage the phy?

gkasprow, 2020-11-09:
I ordered the D1213A-04V-7 TVS that has the same footprint. I want to install them and see if I manage to kill the PHY

sbourdeauducq, 2021-02-10:
@gkasprow Did you do the test, and should we just replace the TVS with D1213A-04V-7 on the existing 2.0 boards to fix this problem?

gkasprow, 2021-02-10:
Yes, I replaced the TVS. I tried hard but didn't manage to kill it.
I have two other designs with identical PHY and PoE circuits, also replaced the diodes and so far no issues.

gkasprow, 2021-03-15:
I think we can close it.

closed 2021-03-15 by gkasprow