sinara-hw/Thermostat #84: Thermostat v2.2 changes

gkasprow, 2020-10-31:
The new release will have the following changes:
- replace TEC inductors with ones having lower losses
- add fan
- add fan fixing holes to the enclosure
- add PWM support for the fan
- add PHY protection resistors and better-suited TVS

hartytp, 2020-11-02:
Track any other improvements from the EEM version.
- [ ] improvements to PoE module layout for thermal performance

dtcallcock, 2020-11-21:
On v2.0 the BOOT0 jumper (JP2) is not labelled on the silkscreen.  Not a huge deal but annoying to have to go through the schematic and figure it out.

jbqubit, 2020-12-06:
- [ ] Fix channel-crossed SENS inputs.

jordens, 2020-12-08:
Pleas don't change that as long as it introduces incompatibilities and as long as there are unaddressed impacts on the firmware.

gkasprow, 2020-12-08:
It can be fixed in firmware already

jordens, 2020-12-08:
Exactly.

gkasprow, 2021-03-15:
PoE thermal issues seem to be solved by a fan and a heatsink on top of the inductors which redirect part of the flow to the PoE module. I described it in #93

gkasprow, 2021-03-15:
added heatsinks 
[image]

closed 2021-03-15 by gkasprow