sinara-hw/Thermostat #86: hardware version pins

jbqubit, 2020-12-04:
@gkasprow says that the hardware version number is electrically readable via hardwired pins. Which pins? Add pin to version mapping to wiki. This will ease implementation of firmware readout.

gkasprow, 2020-12-04:
[image]

hartytp, 2020-12-05:
@jbqubit this is not the kind of info that should be in the wiki imho, particularly since it’s clearly labelled in the schematic. The wiki should be reserved for high level info suitable for users who want to get an idea of the boards capabilities not low-level details like this (otherwise the wiki just ends up duplicating the schematic in a less helpful way). Anyone who can’t figure this out quickly from the schematic is unlikely to be able to do something useful with it in the firmware imho.

closed 2020-12-05 by hartytp