sinara-hw/Thermostat #87: Flashing firmware

jbqubit, 2020-12-04:
Wiki lacks instructions for flashing firmware. From conversation with @gkasprow today.

On a Windows machine install [st.com](https://st.com) DfuSe USB device firmware upgrade (DFU) software. [link](https://www.st.com/en/development-tools/stsw-stm32080.html).  

- [build](https://git.m-labs.hk/M-Labs/thermostat/#user-content-building) firmware .elf 
- locate .elf here: ```target/thumbv7em-none-eabihf/release/thermostat```
- add jumper to Thermostat v2.0 across 2-pin jumper adjacent to JTAG connector
- cycle board power to put it in DFU update mode
- connect micro-USB to PC
- use DFU software to upload .elf to Thermostat uP
- remove jumper
- cycle power to leave DFU update mode

Comments anyone?

gkasprow, 2020-12-04:
You need to convert .elf to DFU firs

jbqubit, 2020-12-04:
I was able to directly use the ```thermostat.elf``` you gave me this afternoon.

Jemetzner, 2020-12-04:
You can also use [dfu-util ](http://dfu-util.sourceforge.net/) to upload a .bin file, which can be found with the release of the latest [builds](https://nixbld.m-labs.hk/job/stm32/stm32/thermostat).

hartytp, 2020-12-05:
IMHO this should be in the firmware repo not the hw repo.

closed 2020-12-05 by hartytp

jbqubit, 2020-12-08:
@hartytp https://git.m-labs.hk/M-Labs/thermostat/pulls/20

hartytp, 2020-12-08:
why are you tagging me?