sinara-hw/Thermostat #92: D12 Zenner diode draws too much current

maciejprzybysz, 2021-02-03:
During thermal tests of the Thermostat, we noticed that the 1SMA5914BT3G Zenner diode (D12) gets warm. The photo from the thermal imaging camera also shows that significant amount of power is emitted on the R35 resistor.
[image]

Measurement with a multimeter shows that, depending on the device (several pieces checked), the R35 resistor drops voltage to ~ 3.15-3.30V. The same voltage was measured at diode D12.
[image]

After desoldering the diode and examining it, it turned out that at 3.15V the diode current is over 50mA. In the [datasheet](https://pl.mouser.com/datasheet/2/308/1SMA5913BT3_D-1101349.pdf) you can find this specification:
[image]

This Zenner should be changed to a more suitable one. The absolute maximum rating on the AD7172 inputs is 3.3V+0.3V so Vz(max)<3.6V and current at 3V3 should be much less than 50mA. I checked distributors stock and didn't find such diode in SMA case (same as 1SMA5914BT3G), so we ordered this one for testing:
[PLZ3V3B-G3-H](https://pl.mouser.com/ProductDetail/Vishay-Semiconductors/PLZ3V3B-G3-H?qs=ECv811tYSNiXaJv05vovrQ%3D%3D)

It should be solderable on existing boards, but changing footprint in the next review should be considered.

gkasprow, 2021-02-03:
This issue was fixed [ already ](https://github.com/sinara-hw/Thermostat/issues/48)

maciejprzybysz, 2021-02-04:
> This issue was fixed [ already ](https://github.com/sinara-hw/Thermostat/issues/48)

Ok, I had to miss this issue. This change is not pushed on latest commit nor is it listed on [v2.2 changes issue](https://github.com/sinara-hw/Thermostat/issues/84), so I thought there was still a problem.

gkasprow, 2021-02-04:
This is still a work in progress

gkasprow, 2021-02-04:
I pushed all the recent changes

gkasprow, 2021-03-15:
fixed

closed 2021-03-15 by gkasprow