sinara-hw/Thermostat #93: Thermostat v2.0 thermal investigation

maciejprzybysz, 2021-02-03:
Due to several github issues regarding thermostat heating problems, we did some tests of devices that we have in Technosystem. You can find full report in the attachment. Hope this helps to get the problems under control.

[Thermostat thermal tests v1.0.pdf](https://github.com/sinara-hw/Thermostat/files/5918715/Thermostat.thermal.tests.v1.0.pdf)

sbourdeauducq, 2021-02-03:
Ouch, 99C is out of spec for the MAX1968, it needs to be <85C according to the datasheet.
The datasheet also says "The thermally enhanced TSSOP-EP package with exposed metal pad minimizes  operating  junction  temperature". It seems that there could be a larger PCB plane connected to these that would spread heat better:
[image]

How to assemble the suggested heat transfer pad considering that the PCB slides into the aluminum housing? Maybe this version of the board shouldn't be used in the small standalone aluminum enclosure, and we should mount some heatsink on the TEC drivers...

sbourdeauducq, 2021-02-03:
Hmm could be that some planes are missing in the Gerber...

maciejprzybysz, 2021-02-03:
We ordered some heat transfer pads and we will try to mount it with aluminum housing. For now we are waiting for delivery. I'll let you know as soon as we check it.

maciejprzybysz, 2021-02-03:
> Hmm could be that some planes are missing in the Gerber...

Hmm, I checked gerbers in last release, but It seems that all planes are on the right place (comparing to pbcdoc). Could you please specify more precisely what you meant by missing planes please?

sbourdeauducq, 2021-02-03:
Does the Gerber screenshot I posted above accurately represent all board layers?

gkasprow, 2021-02-03:
Guys, I designed an enclosure with a fan. I also did initial tests with the fan and all temperatures dropped significantly to a safe level.
All the test report is [here](https://github.com/sinara-hw/Thermostat/issues/73)

gkasprow, 2021-02-03:
@sbourdeauducq the PCB is 4-layer. One is negative so you don't see it.

hartytp, 2021-02-03:
@maciejprzybysz thanks for the nice write up!

maciejprzybysz, 2021-02-05:
We just made some quick test with heat transfer pad between TEC drivers and enclosure. It is not an exact measurement, but it does show how much you can gain with a heat transfer spacer. I think it is worth applying this to v2.0 devices without active cooling.

[Thermostat v2.0 TEC drivers - passive cooling test](https://www.youtube.com/watch?v=f-j0fX6Do9g)

Notice:  
The video starts after 45 minutes of continuous operation with max load on one channel. When the device is removed from the housing, the driver is still working, causing additional heating. After ~15 seconds after housing removal, the thermal imaging camera shows the driver temperature of approx. 73C (and rising).

2mm thermal pad was used - [datasheet](https://www.tme.eu/Document/a9dfa0f7d5e897fc961573a1af067b5f/SP_6.0_en%20%281%29.pdf)

gkasprow, 2021-02-05:
I had a similar experience, but to be more precise I used PT1000 sensors glued to various parts.
However, I discovered that that does not solve the issue with PoE losses and this is the main cause of power shutdowns.
That's why I introduced a small fan.

sbourdeauducq, 2021-02-10:
Greg reported excessive inductor losses in https://github.com/sinara-hw/Thermostat/issues/73#issuecomment-719835089. Is this not a problem anymore?

gkasprow, 2021-02-10:
I'm not sure it was an inductor - after I added a fan and heatsink it's not an issue anymore

gkasprow, 2021-03-15:
I think we can close it.

closed 2021-03-15 by gkasprow