sinara-hw/Thermostat #95: low losses PoE

gkasprow, 2021-02-15:
There are higher efficiency PoE [modules ](http://silvertel.com/images/datasheets/Ag5400-datasheet-high%20Efficiency-30W-Power-Over-Ethernet-Plus-Module-PoE+PD.pdf)available. They have the same pinout. We can use them to limit losses.

jordens, 2021-02-15:
At that point the rectifier diodes will limit efficiency as well.

gkasprow, 2021-02-15:
sure, but they work at 50V so the losses are much lower than the existing PoE module where the output rectification diode works at 12V

jordens, 2021-02-15:
Ah, yes. I miscounted the zeros. We loose about 2% in the diodes.

gkasprow, 2021-02-15:
The difference between AG5300 and AG5400 is that the latter uses the synchronous rectification

gkasprow, 2021-03-15:
The fan solves issues with PoE, I'm not sure if it's worth replacing the PoE module. New issues may appear.

closed 2021-03-15 by gkasprow