sinara-hw/Thermostat #97: Thermostat 2.1 changelog

gkasprow, 2021-03-15:
- replaced EOL buck converter #88 
- replaced EOL TVS #91 
- added SHDN resistor #94 
- fixed DAC PN issue #78 
- fixed VBUS label  #89 
- JTAG pin 1 dot #76 
- replaced TVS; added series resistors #83 
- replaced Zener diode with 1SMA5916BT3G #92

gkasprow, 2021-03-15:
- moved TACHO input to TIMER8 IC @jordens please confirm it's OK
- added PWM fan control, routed to TIMER8 CH4 OC @jordens please confirm it's OK

jordens, 2021-03-15:
Looks good to me

gkasprow, 2021-03-15:
- changed HW_REV

gkasprow, 2021-03-15:
- added test point names

gkasprow, 2021-05-02:
- swapped TEC labels #90 on schematic and front panel

gkasprow, 2021-05-02:
- added fan and screws to the BOM

dtcallcock, 2021-10-11:
What's the timeline for 2.2 release @gkasprow ?  Our 2.0s are working nicely but would be good to have some of the issues fixed for our next order.

closed 2024-06-17 by maciejprzybysz