sinara-hw/Thermostat #116: CPU migration to stm32h7

jordens, 2023-02-22:
@gkasprow could you check whether we can drop in the stm32h7 we have on thermostat_eem and stabilizer? That would reduce the code footprint, simplify maintenance and also enable more population variants/supply chain workarounds.

It should be doable without too much hassle:

https://www.st.com/en/microcontrollers-microprocessors/stm32h7-series.html :

> The STM32H7 single-core and value lines are pin-to-pin compatible with the STM32F7 series of very high-performance MCUs and STM32F4 series of high-performance MCUs for the most common packages (*).

https://www.st.com/content/ccc/resource/technical/document/application_note/73/76/21/47/ef/d0/4c/16/DM00164538.pdf/files/DM00164538.pdf/jcr:content/translations/en.DM00164538.pdf
https://www.st.com/resource/en/application_note/dm00337702-migration-of-microcontroller-applications-from-stm32f7-series-to-stm32h743753-line-stmicroelectronics.pdf

gkasprow, 2023-03-10:
This particular chip has a different placement of ADC channels, but from hw perspective, it's fine
they also differ by the placement of SPI NSS pins but for SPI peripherals we don't use them here

jordens, 2023-03-11:
IIRC there may also be some minor jumpering needed around the analog digital suplies and references.