sinara-hw/Thermostat #118: Add diode to prevent powering Zotino and Rack from PoE

jordens, 2023-03-20:
J1/J3 will happily sink power from PoE
Add a (overridable) ideal diode as on Thermostat_EEM

maciejprzybysz, 2023-06-26:
@jordens Could you please describe what exactly is the problem?
As far as I understand this issue, you want to cut off powering Zotino by PoE thorough J1/J3 connector?
Do we need to disable powering Thermostat from Zotino also?
How do we want to switch P12V on J1/J3 (CPU? jumper?)?

maciejprzybysz, 2023-06-26:
> Same as on Thermostat_EEM.

Thermostat EEM do not have connectors for Zotino with P12V as far as I know. I assume you mean P12V_EEM circuit?

[image]

This circuit is useless in case of switching on/off EEM power supply for Thermostat_EEM. If P12V0_EEM power is present T9 mosfet will be on, no matter what is Force_EEM_Source state. (this should be an issue, shouldn't be?)

> Maybe even simpler/better add a jumper between the RJ45 and the PoE module to disable PoE when connected zo Zotino (even if the PoE module is mounted). That would also solve the issue.

Maybe you could simply turn PoE off on switch config? This wouldn't require Sinara hardware access to modify settings.

jordens, 2023-06-26:
No. That's the intention. It's a diode with an override.

Placing more demand and restrictions on external components makes things harder to use. It's a bad idea.
This is not some "setting": When you connect Zotino, you already have hardware access!

maciejprzybysz, 2024-06-12:
I'd like to make new revision of Thermostat, but I would like to take a broader view first.

Could someone describe use cases of Thermostat? As I understand there are at least 2 scenarios:
1. Standalone version in aluminum case, powered from PoE or front panel 12DC connector, used with 5 pin  J4/J5 backpanel terminal block connectors.
2. 4HP version fitted into Sinara crate, used with Zotino through 10 pin J1/J3 connectors, powered from ........? Molex 4pin or PoE?

Any other use cases?

Ad. 2: 
If we use Thermostat with Zotino, what P12V on J1/J3 is used for? As I understand Zotino should be powered from EEM connector in this case, not from Thermostat. Maybe it could be used to power Thermostat from Zotino?

Bonus question:
What is T1 used for? It is described as "reverse protection", but should It be understand as "reverse polarity protection" (It will not work that way) or to cut off PoE output from P12V when not used (this makes some sense if we really use POE_PWR_SRC signal or if AG5300 could not accept any voltage on output when not used, but description is misleading then)

[image]

@jordens @gkasprow

maciejprzybysz, 2024-08-08:
For now, due to space restriction on PCB and no precise info about use-cases, I just added two 0R jumpers to cut off P12V on J1&J13 connectors to Zotino. This would separate P12V domains if needed.

closed 2024-08-08 by maciejprzybysz