sinara-hw/Thermostat #121: add LICENSE file

jordens, 2024-04-24:
@gkasprow Could you add the CERN OHL 1.2 license as a file, e.g. from https://github.com/sinara-hw/Thermostat_EEM/blob/master/LICENSE.CERN-OHL-v1.2 , to this repo?

gkasprow, 2024-04-24:
And v2 comes in 3 flavours
- A strongly-reciprocal variant, CERN-OHL-S ([txt](https://ohwr.org/cern_ohl_s_v2.txt), [pdf](https://ohwr.org/cern_ohl_s_v2.pdf)). User guide ([txt](/project/cernohl/wikis/uploads/b88fd806c337866bff655f2506f23d37/cern_ohl_s_v2_user_guide.txt), [pdf](/project/cernohl/wikis/uploads/cf37727497ca2b5295a7ab83a40fcf5a/cern_ohl_s_v2_user_guide.pdf)).

- A weakly-reciprocal variant, CERN-OHL-W ([txt](https://ohwr.org/cern_ohl_w_v2.txt), [pdf](https://ohwr.org/cern_ohl_w_v2.pdf)). User guide ([odt](/project/cernohl/wikis/uploads/eb5fac4e02180da7a4d15f99ab48ab7c/cern_ohl_w_v2_howto.odt), [pdf](/project/cernohl/wikis/uploads/c2e5e9d297949b5c2d324a6cbf6adda0/cern_ohl_w_v2_howto.pdf)).

- A permissive variant, CERN-OHL-P ([txt](https://ohwr.org/cern_ohl_p_v2.txt), [pdf](https://ohwr.org/cern_ohl_p_v2.pdf)). User guide ([odt](/project/cernohl/wikis/uploads/f123aac388675e12b308de0ade1a0278/cern_ohl_p_v2_howto.odt), [pdf](/project/cernohl/wikis/uploads/8a6b5d01f71c207c49493e4d114d61e6/cern_ohl_p_v2_howto.pdf)).

jordens, 2024-04-24:
The current data in the repository and the release files say CERN OHL v1.2. Let's add that first, then change license later after deciding how and where to.

jordens, 2024-04-24:
Thanks! Convention appears to be to call them `LICENSE` or `LICENSE.NAME` but I guess it'll be fine as it is.

closed 2024-04-24 by jordens